linrunner/TLP #626: [Thinkpad X230 Coreboot] Recalibration not working

cosmicraccoon, 2022-03-14:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
When trying to recalibrate my `BAT0` it wont let me, using `sudo tlp recalibrate` command:  
```
Setting temporary charge thresholds for BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Error: battery discharge/recalibrate not available.
Battery recalibration aborted.
```  
Information about battery via `sudo tlp-stat -b`:

```
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = inactive (none)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 45N1029
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
```
So the problem seems to be the `tpacpi-bat (acpi_call) = inactive (none)` line, and I just cant get tlp to manage the battery. I just get like 5h on the 9 cell battery when doing light work (coding in nvim).  
Full output of `tlp-stat`: https://pastebin.com/47n6nLTq

**Expected behavior**
Recalibration should work with my battery and also I had much more battery time (around 12-16h) on Void Linux with TLP.

**To Reproduce**
Run `sudo tlp recalibrate`.
Full output of `tlp-stat`: https://gist.github.com/cosmicraccoon/a0398c612317ca40095b91497352ce97

**Additional context**
- Arch Linux
- Kernel: `5.16.14-zen1-1-zen`
- tlp -> `1.5.0`
- acpi_call-dkms -> `1.2.2-1`
- Original 44++ battery
- skullz BIOS from 1vyrain

linrunner, 2022-03-16:
Hi,

are you shure that Coreboot supports force-discharge? Maybe it doesn't.

It also does not seem to provide complete battery data to the kernel, otherwise your `tlp-stat` output should contain data like this:

> /sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
> /sys/class/power_supply/BAT0/energy_full                    =  56160 [mWh]
> /sys/class/power_supply/BAT0/energy_now                     =  35620 [mWh]
> /sys/class/power_supply/BAT0/power_now                      =      0 [mW]
> /sys/class/power_supply/BAT0/status                         = Not charging

Please show the output of

     sudo /usr/share/tlp/tpacpi-bat -v -g FD 1

and

     grep . /sys/class/power_supply/BAT0/*

cosmicraccoon, 2022-03-16:
That was actually my last thought. Maybe coreboot does not support it. But the thing is that I didn't find any similar issues with coreboot.
Output of ` sudo /usr/share/tlp/tpacpi-bat -v -g FD 1`:  
```
-> sudo /usr/share/tlp/tpacpi-bat -v -g FD 1
Error: AE_NOT_FOUND for ASL base: \_SB.PCI0.LPCB.EC.HKEY
Error: AE_NOT_FOUND at /usr/share/tlp/tpacpi-bat line 425.
```
Output of `grep . /sys/class/power_supply/BAT0/*`:
```
-> grep . /sys/class/power_supply/BAT0/*
/sys/class/power_supply/BAT0/alarm:0
/sys/class/power_supply/BAT0/capacity:31
/sys/class/power_supply/BAT0/capacity_level:Normal
/sys/class/power_supply/BAT0/charge_control_end_threshold:100
/sys/class/power_supply/BAT0/charge_control_start_threshold:96
/sys/class/power_supply/BAT0/charge_full:8971000
/sys/class/power_supply/BAT0/charge_full_design:9324000
/sys/class/power_supply/BAT0/charge_now:2867000
/sys/class/power_supply/BAT0/charge_start_threshold:96
/sys/class/power_supply/BAT0/charge_stop_threshold:100
/sys/class/power_supply/BAT0/current_now:1119000
/sys/class/power_supply/BAT0/cycle_count:0
grep: /sys/class/power_supply/BAT0/device: Is a directory
grep: /sys/class/power_supply/BAT0/hwmon2: Is a directory
/sys/class/power_supply/BAT0/manufacturer:LGC
/sys/class/power_supply/BAT0/model_name:45N1029
grep: /sys/class/power_supply/BAT0/power: Is a directory
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/serial_number:16082
/sys/class/power_supply/BAT0/status:Discharging
grep: /sys/class/power_supply/BAT0/subsystem: Is a directory
/sys/class/power_supply/BAT0/technology:Li-ion
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_NAME=BAT0
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_TYPE=Battery
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_STATUS=Discharging
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_PRESENT=1
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_TECHNOLOGY=Li-ion
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CYCLE_COUNT=0
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11100000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_VOLTAGE_NOW=11146000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CURRENT_NOW=1119000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CHARGE_FULL_DESIGN=9324000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CHARGE_FULL=8971000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CHARGE_NOW=2867000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CAPACITY=31
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CAPACITY_LEVEL=Normal
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_MODEL_NAME=45N1029
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_MANUFACTURER=LGC
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_SERIAL_NUMBER=16082
/sys/class/power_supply/BAT0/voltage_min_design:11100000
/sys/class/power_supply/BAT0/voltage_now:11146000
```

linrunner, 2022-03-16:
> Output of ` sudo /usr/share/tlp/tpacpi-bat -v -g FD 1`:
> Error: AE_NOT_FOUND for ASL base: \_SB.PCI0.LPCB.EC.HKEY
> Error: AE_NOT_FOUND at /usr/share/tlp/tpacpi-bat line 425.

tpacpi-bat cannot determine the entry point for the call. I can't say what exactly goes wrong there, but in any case something is different than with a stock ThinkPad BIOS.

I recommend you just wait for kernel 5.17. There force-discharge is already built in and the detour via tpacpi-bat/acpi_call is unnecessary. If it still does not work, then the support in Coreboot is missing.

> /sys/class/power_supply/BAT0/charge_full:8971000
> /sys/class/power_supply/BAT0/charge_full_design:9324000
> /sys/class/power_supply/BAT0/charge_now:2867000
> /sys/class/power_supply/BAT0/current_now:1119000

_charge_full/full_design/now_ sysfiles normally contain values in mAh. But your values are obviously in mWh (if you omit 000 at the end), they fit a ThinkPad 9 cell battery.  _current_now_ is mA too, but the value seems in mW.

So Coreboot delivers wrong values and behaves differently than a stock ThinkPad BIOS, where the sysfiles are always called _energy__* (see my post). In short, this all looks pretty buggy.

> Also TLP doesn't really "manage" the battery life cause I just have like 5h of battery life with the 9-Cell battery.

Since you are talking about battery operation, please show to the output of 

    sudo tlp-stat 

in battery mode, so that I can see if the settings apply.

linrunner, 2022-04-07:
@moonxraccoon : 

1. Kernel 5.17 has landed in Arch. Did you check what happens?
2. You didn't further elaborate on your "management of battery life" issue. May I consider that being no longer relevant?

linrunner, 2022-04-19:
Due to inactivity I'll close here. Nevertheless, you are welcome to post new findings here.

closed 2022-04-19 by linrunner

moonxraccoon, 2022-04-22:
Sorry I forgot to respond.
So basically from what I have found, coreboot does not support battery recalibration.
So I need to go back to the stock BIOS first.

apvs, 2024-01-24:
I was recently digging through the old Coreboot wiki and found some [workaround](https://www.coreboot.org/Board:lenovo/x220#Recalibrate_batteries) for this and similar issues. At least this works for my X220:
```
sudo ./ectool -w 0xb4 -z 0x06
Writing ec b4 = 06
```

The battery immediately started discharging (AC adapter plugged in), tlp-stat shows this:
```
# tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
* tp-smapi (tp_smapi)     = readonly (status)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = SANYO
/sys/devices/platform/smapi/BAT0/model                      = 45N1023
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2023-03-25
/sys/devices/platform/smapi/BAT0/first_use_date             = 2012-03-13
/sys/devices/platform/smapi/BAT0/cycle_count                =     40
/sys/devices/platform/smapi/BAT0/temperature                =     35 [°C]
/sys/devices/platform/smapi/BAT0/design_capacity            =  47520 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  49720 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  18950 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     38 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now =    113 [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    = not_charging [min]
/sys/devices/platform/smapi/BAT0/power_now                  =  -9864 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  = -13882 [mW]
/sys/devices/platform/smapi/BAT0/state                      = discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     75 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto]

Charge                                                      =   38.1 [%]
Capacity                                                    =  104.6 [%]
```

Don't mind the `manufacture_date`/`first_use_date` mess - AFAIK this is typical for some cheap aftermarket batteries.

I'm not sure if this can somehow be implemented in TLP without rewriting the Coreboot code, but perhaps this workaround could be useful to someone with similar problems.

linrunner, 2024-01-25:
Oh my, not yet another external tool (and not another workaround) for ThinkPads :-(  

This does not belong in TLP. Coreboot must be exteneded to provide the ACPI calls BDSG and BDSS that the kernel expects:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/thinkpad_acpi.c#n9337

apvs, 2024-01-25:
Sure, I get it. It looks like the Coreboot devs knew about the problem: https://ticket.coreboot.org/issues/156, but settled on implementing thresholds logic and left it at that, I'll see what I can do about it. Thanks for clarification :)