linrunner/TLP #630: Add support for new AMD P-state performance scaling driver

eternal-sorrow, 2022-04-12:
**Is your feature request related to a problem? Please describe.**

Linux 5.17 was released with a new AMD P-state driver.

**Describe the solution you'd like**

Add support of this driver in TLP.

**Describe alternatives you've considered**

There is a possibility that it does not add anything on top of the current acpi-cpufrec driver, so thi is unneeded. I don't know about that.

**Additional context**

[Kernel documentation](https://docs.kernel.org/admin-guide/pm/amd-pstate.html)

linrunner, 2022-04-19:
Hi, 

the linked specification is extremely scarce. I can neither see what is to be controlled here how, nor which value ranges the attributes have and if they are rw or ro. 

Since I don't own AMD hardware, I can't implement anything here without _much_ more input and sample output from your side.

eternal-sorrow, 2022-04-19:
Okay, what do you need me to do?

linrunner, 2022-04-19:
Let's begin with:

    ls -Rl /sys/devices/system/cpu/
    grep . $(find /sys/devices/system/cpu/ -type f)

**Since this is going to be a long output, please post it via https://gist.github.com/**

linrunner, 2022-04-19:
Also 

      sudo tlp-stat -s -p

please.

eternal-sorrow, 2022-04-19:
I'm having problems trying to do that. I have `amd-pstate` built as a module. I tried to load this module with `modprobe amd-pstate`, and the command did not print any errors, but module does not show up in `lsmod` and `rmmod amd-pstate` prints "rmmod: ERROR: Module amd_pstate is not currently loaded`. I can still upload output of these commands, but I don't think it will be useful.

adomixaszvers, 2022-05-21:
Output of the commands on a machine with Ryzen 7 Pro 5850U https://gist.github.com/adomixaszvers/62fc4dd221de463f5b7d101db7328268

tribals, 2022-06-29:
@linrunner Hello!

I'm interested in advancing this feature. I can report any needed information and do tests. I have Ryzen 3-based laptop.

Cold you please provide some guidance? What needs to be done for this to work?

linrunner, 2022-07-01:
@tribals Fortunately, an AMD ThinkPad  has found its way to my premises in the meantime. I'll be happy to get back to you as soon as there's something to test.

tribals, 2022-07-01:
Great news! 🙂

harrykipper, 2022-07-16:
afaik amd-pstate uses the same governors as acpi-cpufreq. I guess there's not much to change from tlp's perspective.

linrunner, 2022-08-12:
Hi,

from what I read in the [kernel docs]([url](https://docs.kernel.org/admin-guide/pm/amd-pstate.html#user-space-interface-in-sysfs)) there are only r/o attributes. I've added them to the tlp-stat -p output.

@harrykipper @adomixaszvers @eternal-sorrow @tribals Please test with the main branch and produce the output of

    tlp-stat -s -p

Thanks in advance.

harrykipper, 2022-08-12:
```
--- TLP 1.6.0-alpha.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad E14 Gen 3 20YECTO1WW
BIOS           = R1OET34W (1.13 )
OS Release     = Arch Linux
Kernel         = 5.19.0-S #1 SMP PREEMPT Thu Aug 11 11:18:32 BST 2022 x86_64
/proc/cmdline  = root=/dev/nvme0n1p2 resume=/dev/nvme0n1p3 thinkpad_acpi.fan_control=Y zswap.enabled=1 nmi_watchdog=0 drm.vblankoffdelay=1 snd_hda_intel.power_save=1 snd_hda_intel.power_save_controller=Y snd_ac97_codec.power_save=1 drm_kms_helper.poll=0 amdgpu.aspm=1 amdgpu.bapm=1 amdgpu.dpm=1 amdgpu.dc=1 amdgpu.dcfeaturemask=0x1 amdgpu.dcfeaturemask=0x80 amdgpu.dcfeaturemask=0x8 amdgpu.runpm=1 usbcore.autosuspend=2 btusb.enable_autosuspend=Y nvme_core.streams=Y nvme_core.force_apst=Y nvme_core.default_ps_max_latency_us=11276000 iwlmvm.power_scheme=3 amdgpu.ppfeaturemask=0xffffffff amdgpu.deep_color=1 iwlwifi.power_save=Y iwlwifi.power_level=5 iwlwifi.uapsd_disable=0 mitigations=off msr.allow_writes=on cryptomgr.notests no_timer_check noreplace-smp page_alloc.shuffle=1 rcu_nocbs=0-15 rcupdate.rcu_expedited=1 tsc=reliable rootfstype=f2fs
Init system    = systemd 
Boot mode      = UEFI
Sleep mode     = deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 16:35:13,    521 sec(s) ago
Mode           = battery
Power source   = battery

+++ Processor
CPU model      = AMD Ryzen 7 5800U with Radeon Graphics

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = amd-pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = conservative ondemand userspace powersave performance schedutil 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  4507000 [kHz]

/sys/devices/system/cpu/cpu1..cpu15: omitted for clarity, use -v to show all

/sys/devices/system/cpu/cpufreq/boost                  = 1
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_highest_perf          =      166 [%]
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_max_freq              =  4507000 [kHz]
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_lowest_nonlinear_freq =  1112000 [kHz]

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = (not available)

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = low-power
/sys/firmware/acpi/platform_profile_choices            = low-power balanced performance
/sys/devices/platform/thinkpad_acpi/dytc_lapmode       = 0
```

adomixaszvers, 2022-08-12:
```
--- TLP 1.6.0-alpha.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T14 Gen 2a 20XK007DMH
BIOS           = R1MET47W (1.17 )
OS Release     = NixOS 22.05 (Quokka)
Kernel         = 5.18.15 #1-NixOS SMP PREEMPT_DYNAMIC Fri Jul 29 15:28:18 UTC 2022 x86_64
/proc/cmdline  = initrd=\efi\nixos\pldnpfk4ag4bc49fxnhd07q1q1mv7jh2-initrd-linux-5.18.15-initrd.efi init=/nix/store/6wmwnfmq8yrg1i4z8f4n77h63aqrd7lx-nixos-system-adomo-t14-22.05.20220803.478f3cb/init acpi_backlight=native initcall_blacklist=acpi_cpufreq_init nohibernate loglevel=4
Init system    = systemd 
Boot mode      = UEFI
Sleep mode     = s2idle deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 19:56:54,    824 sec(s) ago
Mode           = AC
Power source   = AC

+++ Processor
CPU model      = AMD Ryzen 7 PRO 5850U with Radeon Graphics

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = amd-pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance schedutil 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  4507000 [kHz]

/sys/devices/system/cpu/cpu1..cpu15: omitted for clarity, use -v to show all

/sys/devices/system/cpu/cpufreq/boost                  = 1
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_highest_perf          =      166 [%]
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_max_freq              =  4507000 [kHz]
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_lowest_nonlinear_freq =  1112000 [kHz]

/sys/module/workqueue/parameters/power_efficient       = N
/proc/sys/kernel/nmi_watchdog                          = (not available)

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = balanced
/sys/firmware/acpi/platform_profile_choices            = low-power balanced performance
/sys/devices/platform/thinkpad_acpi/dytc_lapmode       = 0

```

kikislater, 2022-08-12:
```
--- TLP 1.6.0-alpha.0 --------------------------------------------

+++ System Info
System         = HP  HP EliteBook 845 G8 Notebook PC
BIOS           = T82 Ver. 01.10.00
OS Release     = Manjaro Linux
Kernel         = 5.19.0-2-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Aug 2 18:25:43 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/@/boot/vmlinuz-5.19-x86_64 root=UUID=b67b7b03-d01d-46ae-b0bd-b409de2d7195 rw rootflags=subvol=@ quiet apparmor=1 security=apparmor udev.log_priority=3 resume=/dev/disk/by-uuid/aafa4122-12b3-4577-9f3b-413d3fe59733
Init system    = systemd 
Boot mode      = UEFI
Sleep mode     = s2idle

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 22:29:33,     34 sec(s) ago
Mode           = battery
Power source   = battery

+++ Processor
CPU model      = AMD Ryzen 7 PRO 5850U with Radeon Graphics

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = conservative ondemand userspace powersave performance schedutil 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =  1600000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  1900000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies = 1900000 1800000 1600000 [kHz]

/sys/devices/system/cpu/cpu1..cpu15: omitted for clarity, use -v to show all

/sys/devices/system/cpu/cpufreq/boost                  = 1

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = (not available)
/sys/firmware/acpi/platform_profile_choices            = (not available)

```

tribals, 2022-08-13:
```
--- TLP 1.6.0-alpha.0 --------------------------------------------

+++ System Info
System         = LENOVO IdeaPad 5 Pro 16ACH6 82L5
BIOS           = GSCN29WW
OS Release     = Solus 4.3 Fortitude
Kernel         = 5.18.7-2.stable #1 SMP PREEMPT_DYNAMIC Thu Jun 30 21:13:13 UTC 2022 x86_64
/proc/cmdline  = root=UUID=f63e5944-834d-498a-bd59-0a6587ce000e quiet loglevel=3 splash systemd.show_status=false rw radeon.si_support=0 radeon.cik_support=0 amdgpu.si_support=1 amdgpu.cik_support=1 module_blacklist=nouveau,k10temp vfio_pci.ids=10de:25a2
Init system    = systemd v250 (250)
Boot mode      = UEFI
Sleep mode     = s2idle

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 13:02:11,     91 sec(s) ago
Mode           = battery
Power source   = battery

+++ Processor
CPU model      = AMD Ryzen 7 5800H with Radeon Graphics

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = amd-pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = ondemand performance schedutil 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3201000 [kHz]

/sys/devices/system/cpu/cpu1..cpu15: omitted for clarity, use -v to show all

/sys/devices/system/cpu/cpufreq/boost                  = 0
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_highest_perf          =      166 [%]
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_max_freq              =  4463000 [kHz]
/sys/devices/system/cpu/cpufreq/policy0/amd_pstate_lowest_nonlinear_freq =  1100000 [kHz]

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = balanced
/sys/firmware/acpi/platform_profile_choices            = low-power balanced performance
```

linrunner, 2022-08-13:
Thank you all.

@kikislater : you may want to use the kernel boot option `amd_pstate.enable=1` to activate the _amd_pstate_ driver.

kikislater, 2022-08-15:
even with this option, I couldn't enbale it ! 

```
--- TLP 1.6.0-alpha.0 --------------------------------------------

+++ System Info
System         = HP  HP EliteBook 845 G8 Notebook PC
BIOS           = T82 Ver. 01.10.00
OS Release     = Manjaro Linux
Kernel         = 5.19.0-2-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Aug 2 18:25:43 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/@/boot/vmlinuz-5.19-x86_64 root=UUID=b67b7b03-d01d-46ae-b0bd-b409de2d7195 rw rootflags=subvol=@ amd_pstate.enable=1 iommu=pt quiet apparmor=1 security=apparmor udev.log_priority=3 resume=/dev/disk/by-uuid/aafa4122-12b3-4577-9f3b-413d3fe59733
Init system    = systemd 
Boot mode      = UEFI
Sleep mode     = s2idle

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 09:44:17,     22 sec(s) ago
Mode           = battery
Power source   = battery

+++ Processor
CPU model      = AMD Ryzen 7 PRO 5850U with Radeon Graphics

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = conservative ondemand userspace powersave performance schedutil 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =  1600000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  1900000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies = 1900000 1800000 1600000 [kHz]

/sys/devices/system/cpu/cpu1..cpu15: omitted for clarity, use -v to show all

/sys/devices/system/cpu/cpufreq/boost                  = 1

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = (not available)
/sys/firmware/acpi/platform_profile_choices            = (not available)

```

amorozov, 2022-08-21:
Is usage of TLP from master really required? I have 1.5.0 installed (Manjaro) 

I've got amd_pstate working and actually can change power profiles using `cpupower`. If 1.5.0 is enough I can provide required outputs in different modes with different governors and frequency limits

linrunner, 2022-08-22:
@amorozov : 

Yes, there is nothing to parametrize on _amd_pstate_ so far that TLP couldn't already do with 1.5. 

But no, I added some of _amd_pstate_'s readonly parameters to `tlp-stat`'s output.  This extension I wanted to have tested with themMain branch. In the meantime I have enough results in the meantime, so you don't have to bother.

linrunner, 2022-09-15:
@harrykipper : that was to be expected. Thanks for the heads up.

brotfessor, 2022-10-26:
I am currently already using the amd_pstate-epp patches metioned above ([PATCH v2](https://lore.kernel.org/lkml/20221009071033.21170-1-Perry.Yuan@amd.com/) atm) and the sysfs interface seems to be exactly the same as with intel, so supporting this seems relatively straightforward.

I am guessing this has to wait until it is accepted in the kernel, but if anyone wants to test it out, here is a quick patch adding support for TLP and it seems to work as expected on my Ryzen 5825U.

Patch: https://github.com/brotfessor/TLP/commit/add1819c67b1a26fe23c9ad36d29d619aae39312

linrunner, 2022-10-26:
@brotfessor : interesting news. Please provide a pull request as soon as its accepted in the kernel. Thank you.

linrunner, 2022-12-04:
[Patch v4](https://lore.kernel.org/lkml/20221110175847.3098728-1-Perry.Yuan@amd.com/)

linrunner, 2022-12-08:
Here we go again: [Patch v7](https://lore.kernel.org/lkml/20221208111852.386731-1-perry.yuan@amd.com/)

linrunner, 2022-12-18:
@harrykipper @brotfessor @amorozov and @all: I finally took the time to build a kernel with [patch v7](https://lore.kernel.org/lkml/20221208111852.386731-1-perry.yuan@amd.com/) and completed the implementation in TLP:

> - CPU_ENERGY_PERF_POLICY_ON_AC/BAT,
>   CPU_HWP_DYN_BOOST_ON_AC/BAT: support AMD Zen 2 or newer CPUs
>   *EXPERIMENTAL* requires amd_pstate scaling driver in active mode
>   provided by a yet unreleased kernel 6.x

You're invited to test with my main branch and provide the output of

    tlp-stat -s -p

ps. amd_pstate active mode requires kernel cmdline option `amd_pstate=active`

My outputs: https://gist.github.com/linrunner/28cafe1fa3bfacc44630d7b5600c0306

amorozov, 2022-12-18:
Thanks, I'll give it a try as soon as I can.

brotfessor, 2022-12-18:
Thank you, I just tried it out, seems to work fine with patch v7:

My output: https://gist.github.com/brotfessor/5f60fe2f079981f6a8f4e7914ca56415

linrunner, 2022-12-19:
For me suspend and reboot hang with the patched kernel. I would not recommend the driver for production use yet.

@amorozov try [this one](https://gist.github.com/linrunner/49b73e96870c9dcff8905632abe546ad) (without any support)

linrunner, 2022-12-19:
Btw: _amd_pstate_ (in active mode) has no option to control frequency boost yet, see [Perrys answer](https://lore.kernel.org/lkml/DM4PR12MB527891F1ABD8226CEB424BBD9CE59@DM4PR12MB5278.namprd12.prod.outlook.com/). This means for TLP that CPU_BOOST_ON_AC/BAT has no function.

harrykipper, 2022-12-19:
Hi, using v8 (https://lore.kernel.org/lkml/20221219064042.661122-1-perry.yuan@amd.com/T/#md63cce4f471ea449904f5f87307cfdddb262e6e4), output's here

https://gist.github.com/harrykipper/96d352e6cfb2f72b67790d5579b73ae4

linrunner, 2022-12-19:
> Notice that the boost bit has now been removed.
This was to be expected: https://lore.kernel.org/lkml/Y5b%2FIvbeIq4d56nd@amd.com/

Btw: there is currently no frequency boost either: https://lore.kernel.org/lkml/DM4PR12MB527891F1ABD8226CEB424BBD9CE59@DM4PR12MB5278.namprd12.prod.outlook.com/

amorozov, 2022-12-25:
Mine (6.1.1 v8) is https://gist.github.com/amorozov/c817354c94338e03c253d96c62ee76de

BTW, the power consumption on 6.x kernel *seems* to be slightly higher than ≈6 months ago (about 1W, 6W+ instead of 5W+) with no notable changes in the desktop environment and applications set. Did anyone else experience anything like that?

linrunner, 2023-02-01:
The patchset seems to be nearing completion https://lore.kernel.org/all/20230131090016.3970625-1-perry.yuan@amd.com/

linrunner, 2023-02-08:
v12 queued for 6.3: https://lore.kernel.org/all/CAJZ5v0gks1z-GkS=3oY1sn0N-sjK_Vh9bXgygqoE71hnKU5B2w@mail.gmail.com/

linrunner, 2023-02-13:
TLP fixed for v12.

brotfessor, 2023-02-20:
Thanks, it is working fine for me on Linux-6.2 with the branch "bleeding-edge" from linux-pm merged:
https://gist.github.com/brotfessor/7b794a2cb612b12a18e57a6c36996d32

FYI, regarding suspend issues, I had it working fine on v6.1 with PATCH v7 manually applied, but v6.2 with the current PATCH v12 occasionally had resume failures for me where the integrated graphics wouldn't come back up and fail resume with a "Error waiting for DMUB idle". I couldn't pinpoint the problem, but since what is to be merged into v6.3 seems to work, despite no substantial differences to the bare patchset, it is probably pointless to spend more time on this.

But I also found out that the error messages and probability of it happening changed significantly with a amdgpu firmware update (mine was from 2021 before that), so if anyone runs into a similar problem again, maybe try updating that.

harrykipper, 2023-02-20:
Resume issues seem to be all over the place for Ryzen 5000 series. See i.e. here https://gitlab.freedesktop.org/drm/amd/-/issues/2403

eternal-sorrow, 2023-04-18:
Why could it be like this:
```
tlp-stat -p        
--- TLP 1.5.0 --------------------------------------------

+++ Processor
CPU model      = AMD Ryzen 7 PRO 5850U with Radeon Graphics

/sys/devices/system/cpu/cpufreq/boost                  = (not available)

/sys/module/workqueue/parameters/power_efficient       = N
/proc/sys/kernel/nmi_watchdog                          = (not available)

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = (not available)
/sys/firmware/acpi/platform_profile_choices            = low-power balanced performance
/sys/devices/platform/thinkpad_acpi/dytc_lapmode       = 1
```

snakeroot, 2023-04-24:
Seems to work fine with just-released kernel 6.3.

linrunner, 2023-04-26:
Works here too.

ysooqe, 2023-05-04:
One quick question, on the latest release (1.5.0), setting `energy_performance_preference` (TLP settings `CPU_ENERGY_PERF_POLICY_ON_AC` / `CPU_ENERGY_PERF_POLICY_ON_BAT`) does not seem to work when using `amd_pstate=active` (`amd_pstate_epp`) on linux 6.3.1.
It can be changed manually, but TLP does not seem to be able to change it on AC or on BATTERY.
Is this already addressed through 5248ad1a4e7076c68b3e581dbb4f26fc4c414495?

linrunner, 2023-05-04:
@ysooqe  _CPU_ENERGY_PERF_POLICY_ON_AC /BAT_ for _amd_pstate_ will be released with 1.6. Feel free to try the alpha packages: https://download.linrunner.de/packages/

ysooqe, 2023-05-04:
> @ysooqe `CPU_ENERGY_PERF_POLICY_ON_AC/BAT` for _amd_pstate_ will be released with 1.6. Feel free to try the alpha packages: https://download.linrunner.de/packages/

Amazing, thank you!

siddhpant, 2023-06-24:
AMD driver also added support for a guided autonomous mode since 6.4.

https://lore.kernel.org/linux-pm/20230307112740.132338-1-wyes.karny@amd.com/

siddhpant, 2023-07-01:
Relevant patches to cpupower: https://lore.kernel.org/lkml/20230619190503.4061-1-wyes.karny@amd.com/

linrunner, 2023-07-14:
@siddhpant to be honest, I don't understand what the difference is between _passive_ and _guided_ in practical use. Can you explain?

@all I've added the new setting `CPU_DRIVER_OPMODE_ON_AC/BAT`. With it may change the operation mode (_active_, _passive_, _guided_) at runtime - i.e. without boot options/reboot - for _intel_pstate_ too. Please test and post your results.

- https://github.com/linrunner/TLP/blob/main/tlp.conf.in#L78
- https://github.com/linrunner/TLP/commit/3e9a3d6324e5efc0ebd7641b8219323b6f058990

siddhpant, 2023-07-14:
> @siddhpant to be honest, I don't understand what the difference is between _passive_ and _guided_ in practical use. Can you explain?

The governor just specifies the range of frequencies in guided mode. In passive mode, the governor dictates the frequencies.

Guided mode apparently leads to better power utilization on average.

In active mode, the `powersave` governor with `powersave` epp can also lead to max frequency. With guided mode, the governor can specify the max frequency. Guided is basically active mode with range enforced, as far as I could understand.

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out and all changes mentioned in this issue -> https://github.com/linrunner/TLP/issues/700

@siddhpant : thanks for your explanation. I will allow myself to use it for the documentation website.

linrunner, 2023-07-24:
@siddhpant : see https://linrunner.de/tlp/settings/processor.html

siddhpant, 2023-07-25:
> @siddhpant : see https://linrunner.de/tlp/settings/processor.html

Perfect. Thank you very much!

linrunner, 2023-07-26:
Post of **@tisyang** copied from https://github.com/linrunner/TLP/issues/700#issuecomment-1650936951

@linrunner 

I found out, if 
```
CPU_DRIVER_OPMODE_ON_AC=guided

CPU_SCALING_GOVERNOR_ON_AC=performance
```
The mode will be `guided` and CPU frequency is fine.

But
```
CPU_DRIVER_OPMODE_ON_AC=guided

CPU_SCALING_GOVERNOR_ON_AC=powersave
```
The mode will be `guided`, but CPU frequency will locked in 400~1000Mhz. This maybe not TLP's bug.

> [tlp-stat report](https://gist.github.com/tisyang/a67baaa5a22bfa58bc82754570456e36)
> 
> My cpu is amd r7-7840hs, kernel 6.4, use 1.6 beta tlp, if choose CPU_DRIVER_OPMODE_ON_AC/BAT = `guided`, the cpu is locked 400 ~ 900 Mhz. But choose `active`, cpu frequency is fine.
> 
> choose `guided` in tlp
> 
> ```
>  sudo cpupower -c 0-15 frequency-info | grep "current CPU" | grep Hz
>   current CPU frequency: 547 MHz (asserted by call to kernel)
>   current CPU frequency: 544 MHz (asserted by call to kernel)
>   current CPU frequency: 549 MHz (asserted by call to kernel)
>   current CPU frequency: 541 MHz (asserted by call to kernel)
>   current CPU frequency: 546 MHz (asserted by call to kernel)
>   current CPU frequency: 549 MHz (asserted by call to kernel)
>   current CPU frequency: 548 MHz (asserted by call to kernel)
>   current CPU frequency: 547 MHz (asserted by call to kernel)
>   current CPU frequency: 548 MHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 547 MHz (asserted by call to kernel)
>   current CPU frequency: 549 MHz (asserted by call to kernel)
>   current CPU frequency: 545 MHz (asserted by call to kernel)
>   current CPU frequency: 545 MHz (asserted by call to kernel)
>   current CPU frequency: 545 MHz (asserted by call to kernel)
> ```
> 
> choose `guided` in tlp
> 
> ```
> sudo cpupower -c 0-15 frequency-info | grep "current CPU" | grep Hz
>   current CPU frequency: 1.31 GHz (asserted by call to kernel)
>   current CPU frequency: 1.27 GHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 3.20 GHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 1.29 GHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 400 MHz (asserted by call to kernel)
>   current CPU frequency: 1.30 GHz (asserted by call to kernel)
>   current CPU frequency: 1.54 GHz (asserted by call to kernel)
>   current CPU frequency: 1.30 GHz (asserted by call to kernel)
>   current CPU frequency: 3.21 GHz (asserted by call to kernel)
>   current CPU frequency: 2.80 GHz (asserted by call to kernel)
> ```
> 
> If set kernel parameter `amd_pstate=guided`, CPU frequency is fine too.

linrunner, 2023-07-26:
@tisyang: right, this is a driver issue. I suggest to file a kernel bug, the driver is quite young and may need improvement.

You maI suggest you try to change to _passive_ and back and see what happens:

    sudo tlp start -- CPU_DRIVER_OPMODE_ON_AC=passive CPU_DRIVER_OPMODE_ON_BAT=passive
    sudo tlp start

siddhpant, 2023-07-29:
Debian testing finally [got 6.4 today](https://tracker.debian.org/news/1448736/linux-signed-amd64-6441-migrated-to-testing/), so I am now able to test guided mode, and switching between modes.

Relevant TLP Settings:

```
CPU_DRIVER_OPMODE_ON_AC=active
CPU_DRIVER_OPMODE_ON_BAT=guided

CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=conservative

CPU_SCALING_MIN_FREQ_ON_BAT=400000
CPU_SCALING_MAX_FREQ_ON_BAT=1800000
```

The mode changes from active to guided and vice-verse successfully upon switching off/on the AC connection.

---

Though one doubt I have is regarding frequency. Running `cpupower frequency-info` yeilds:

```
analyzing CPU 7:
  driver: amd-pstate
  CPUs which run at the same hardware frequency: 7
  CPUs which need to have their frequency coordinated by software: 7
  maximum transition latency: 20.0 us
  hardware limits: 400 MHz - 2.10 GHz
  available cpufreq governors: userspace powersave ondemand conservative performance schedutil
  current policy: frequency should be within 400 MHz and 1.80 GHz.
                  The governor "conservative" may decide which speed to use
                  within this range.
  current CPU frequency: Unable to call hardware
  current CPU frequency: 2.75 GHz (asserted by call to kernel)
  boost state support:
    Supported: yes
    Active: no
    AMD PSTATE Highest Performance: 255. Maximum Frequency: 4.06 GHz.
    AMD PSTATE Nominal Performance: 132. Nominal Frequency: 2.10 GHz.
    AMD PSTATE Lowest Non-linear Performance: 70. Lowest Non-linear Frequency: 1.11 GHz.
    AMD PSTATE Lowest Performance: 26. Lowest Frequency: 400 MHz.
```

Why is the frequency 2.75 GHz? Nominal max is 2.1 GHz, and the output says boost is not active either.

siddhpant, 2023-12-05:
A follow-up to my previous comment:

It seems like that was a problem in the AMD driver. The fix has landed in 6.7.

https://lore.kernel.org/lkml/20231117063839.17465-1-wyes.karny@amd.com/

I was setting up TLP on another laptop and came across that.

linrunner, 2023-12-08:
That is fatal, of course ...