linrunner/TLP #631: Bisecting possible power management issue on ThinkPad X1 Extreme Gen 4

ginkel, 2022-04-25:
Hi there,

I have been successfully using TLP for all of my laptops since 2011. When activating it on my ThinkPad X1 Extreme Gen 4, though, the laptop is having sporadic USB disconnects and sporadically loses keystrokes on the internal keyboard.

While this is certainly not TLP's fault, I am looking for a way to figure out which (power management related) tweak applied by TLP is causing the erratic behavior.

Is there a elegant way to activate the tweaks applied by TLP one-by-one to figure out the cause?

Thanks,
Thilo

linrunner, 2022-05-20:
Due to inactivity I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2022-05-20 by linrunner

ginkel, 2022-05-21:
Thanks for your reply and sorry for not getting back to you earlier! ATM the issue seems to be caused by a Lenovo firmware bug rather than a Linux PM issue.

linrunner, 2022-05-23:
Thanks for your feedback.