linrunner/TLP #633: No mirrors for Fedora 36

qubbus, 2022-05-12:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
No mirrors for Fedora 36
https://repo.linrunner.de/fedora/tlp/mirrors/36/x86_64/tlp.txt

```
tlp RPM packages                                178  B/s | 196  B     00:01    
Errors during downloading metadata for repository 'tlp':
  - Status code: 404 for https://repo.linrunner.de/fedora/tlp/mirrors/36/x86_64/tlp.txt (IP: 85.13.161.171)
Error: Failed to download metadata for repo 'tlp': Cannot prepare internal mirrorlist: Status code: 404 for https://repo.linrunner.de/fedora/tlp/mirrors/36/x86_64/tlp.txt (IP: 85.13.161.171)
tlp RPM packages                                179  B/s | 196  B     00:01    
Errors during downloading metadata for repository 'tlp-updates':
  - Status code: 404 for https://repo.linrunner.de/fedora/tlp/mirrors/36/x86_64/tlp-updates.txt (IP: 85.13.161.171)
Error: Failed to download metadata for repo 'tlp-updates': Cannot prepare internal mirrorlist: Status code: 404 for https://repo.linrunner.de/fedora/tlp/mirrors/36/x86_64/tlp-updates.txt (IP: 85.13.161.171)
Ignoring repositories: tlp, tlp-updates

```
**Expected behavior**

Mirrors should be reacheable

**To Reproduce**

Steps to reproduce the unexpected behavior:

Add repo: dnf install https://repo.linrunner.de/fedora/tlp/repos/releases/tlp-release.fc$(rpm -E %fedora).noarch.rpm and try to install 
akmod-acpi_call akmod-tp_smapi packages

https://repo.linrunner.de/fedora/tlp/mirrors/36/x86_64/tlp.txt

**Additional context**
None

kk0nrad, 2022-05-12:
I was going  to report this myself. I have the same issue.

linrunner, 2022-05-12:
Do you realise that with F36 providing kernel 5.17 you no longer need the kernel module _acpi_call_ (and the whole akmod stuff) for your ThinkPads?

See: https://linrunner.de/tlp/faq/battery.html#which-external-kernel-module-do-i-need-for-my-thinkpad

closed 2022-05-12 by qubbus

linrunner, 2022-05-12:
Installation instructions for Fedora updated.

https://linrunner.de/tlp/installation/fedora.html