linrunner/TLP #634: Can't use threshold feature in Galaxy Book Pro 360

yth1130, 2022-05-14:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Can't use threshold feature.

The plugin part shows generic.

Here is the tlp-stat result [https://gist.github.com/yth1130/033d580dafee031d0913eb30a34ead07](url)

**Expected behavior**

The plugin part shows samsung and threshold function is available.

**To Reproduce**

1. install TLP in Galaxy Book Pro 360, Ubuntu 22.04 by apt(not use ppa)
2. execute systemctl and tlp start.
3. reboot
4. execute tlp-stat

**Additional context**

Maybe the Samsung battery driver is missing, how can I add one?

yth1130, 2022-05-16:
Sorry. I found many people with the same problem and this problem is entirely due to the Samsung battery driver.

closed 2022-05-16 by yth1130

linrunner, 2022-05-16:
Hi, no problem. Thanks for your report anyway.

May I still have some symptom data for future cases? Please show:

    sudo modprobe -v samsung-laptop
    ls -la /sys/devices/platform/samsung

yth1130, 2022-05-16:
Thank you.

> sudo modprobe -v samsung_laptop
libkmod: ERROR ../libkmod/libkmod-config.c:712 kmod_config_parse: /etc/modprobe.d/blacklist.conf line 54: ignoring bad line starting with 'snd-hda-intel'
insmod /lib/modules/5.17.7-051707-generic/kernel/drivers/platform/x86/samsung-laptop.ko 
modprobe: ERROR: could not insert 'samsung_laptop': No such device

And folder "/sys/devices/platform/samsung" is not exist in my laptop.
>ls -a /sys/devices/platform
 .                   INTC1040:00   USBC000:00         power
 ..                  INTC1051:00   coretemp.0         reg-dummy
 ACPI0003:00         MSFT0101:00   efivars.0          regulatory.0
 ACPI000C:00         PNP0C0C:00    eisa.0             rtc-efi.0
 ACPI000E:01         PNP0C0D:00    i8042              serial8250
 ACPI0011:00         PNP0C14:01    intel_rapl_msr.0   snd-soc-dummy
'Fixed MDIO bus.0'   PNP0C14:02    kgdboc             uevent
 INT33A1:00          PNP0C14:03    microcode
 INT34C5:00          SAM0428:00    pcspkr

linrunner, 2022-05-16:
The _samsung_laptop_ kernel module, which is a prerequisite for charge threshold support, does not support your hardware, .