linrunner/TLP #636: tlp disables eth0 when boot on battery

someu0, 2022-05-19:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When I boot laptop while using battery (without connected AC) eth0 interface is in DOWN state.
I can't ping router.

ip link set eth0 down/up brings up connection.

If I boot with AC plugged in eth0 is up after boot.

**Expected behavior**

I expected to have either of those:
1) eth0 interface is up on startup
2) tlp.conf has notion about disabling eth0 interface on boot with battery and option to change this behavior in config file 

**To Reproduce**

1. Problem occures on battery.
2. Actions to reproduce the behaviour
    a) install tlp on laptop connected to LAN router
    b) remove power supply
    c) reboot or poweroff/poweron and check network
4. Shell commands entered and their output
> reboot

> ping 192.168.1.1
ping: connect: Network is unreachable

> ip link show eth0
eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 00 brd ff:ff:ff:ff:ff:ff

6. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

<script src="https://gist.github.com/someu0/cc46b58c735cbdb9bac58d26e6943562.js"></script>

**Additional context**

I installed tlp to save state/unblock on boot wifi.
Without tlp wifi listed as blocked by rfkill.

But after install another problem appeared tlp blocks eth0 on boot with battery.

It took long time to figure out what couses problem because laptop is not always is connected to AC.
It is still possible something else causes the problem and this is a coincidence because I checked this only ~6 times.

linrunner, 2022-05-20:
Hi, 

in general the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html) will help you to narrow down the problem.

In your case I would first try to exclude the ethernet adapter from runtime pm, e.g. enter its address into [RUNTIME_PM_DENYLIST](https://linrunner.de/tlp/settings/runtimepm.html#runtime-pm-denylist).

Background: the default configuration activates runtime pm only in battery mode.

someu0, 2022-05-20:
Configured _RUNTIME_PM_DENYLIST="01:00.0"_
Still no connection on boot without AC

> lspci | grep -w Ethernet
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 10)

I don't see RUNTIME_PM_DENYLIST in tlp-stat output. If it must be there.
https://gist.github.com/someu0/0d4cd897bd42969e024a0876094ff6fd

-----------------
Setting
> RUNTIME_PM_ON_BAT=on 
also had no effect. 

With one of those options either RUNTIME_PM_ON_BAT=on  or  RUNTIME_PM_DENYLIST="01:00.0"
eth0 stays down after _ip link set eth0 down/up_. Only connecting AC enables eth0.

linrunner, 2022-05-26:
> don't see RUNTIME_PM_DENYLIST in tlp-stat output. If it must be there. 

Maybe you didn't remove the leading `#` or you added a `# comment` to your changed parameter lines?

linrunner, 2022-08-26:
Due to inactivity I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2022-08-26 by linrunner