linrunner/TLP #641: Does TLP support Non-systemd init?

djypku, 2022-06-01:
Hi I am working on my own tool to switch to different configure of TLP which works like Power-porfiles-daemon. I want to know if TLP supports Non-systemd init so that I can decide whether I need to provide such support.
Thanks.

closed 2022-06-01 by djypku

linrunner, 2022-06-02:
I assume from your question that you intend to make dynamic changes to the settings by restarting the service. This is the WRONG way to do it.

The CORRECT way to do this is to call `tlp start` (as root)

You should also consider contributing the desired functionality to https://github.com/d4nj1/TLPUI instead of doing everything yourself.

djypku, 2022-06-02:
> I assume from your question that you intend to make dynamic changes to the settings by restarting the service. This is the WRONG way to do it (no matter how often this may be suggested in ill-informed internet forums).
> 
> The CORRECT way to do this is to call `tlp start` (as root).
> 
> The service is only there for system start-up and shutdown.
> 
> You should also consider contributing the desired functionality to [TLPUI](https://github.com/d4nj1/TLPUI) instead of doing everything yourself.

Definitely I know it is wrong way. I just need to start my own daemon to govern  `/usr/sbin/tlp` and I need `systemd` to autostart it. However now I have got how to skip setting up a daemon.
You can see my work at [here](https://github.com/djypku/tlp-switch)
This work is rather bad and need daemon and completely written in shell, but it works fine for my laptop. So instead of revising it I prefer to rewrite it. That may take me some times, but you can wait for it.

linrunner, 2022-06-02:
I see. Sorry for the wrong assumption.