linrunner/TLP #643: Thinkpad P15s Gen2i CPU fans spin up when AC is plugged in, only when using TLP

kpfleming, 2022-07-16:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Since kernel 5.16 was released (and continuing with 5.17 and 5.18), if TLP is enabled on my laptop then when I plug in AC power the CPU fans spin up to about medium speed, even though the CPU temperature has not increased. They continue running at that speed until AC power is disconnected, although their speed will increase as expected if the CPU is loaded and its temperature increases.

**Expected behavior**

Plugging in AC power should not affect CPU fan behavior.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? Only on AC.
2. Install tlp 1.5.0-1 (from Debian Bookworm repository) with configuration in linked gist below, and ensure that TLP_ENABLE is not set to zero.
3. Connect power to one of the laptop's USB-C ports (multiple USB-C power supplies have been tested, as has a Lenovo docking station).
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots) https://gist.github.com/kpfleming/174426ec86f7dbc36a9b07ac2a7312e7

**Additional context**

Interestingly, if the laptop is booted with TLP_ENABLE=0, with AC power connected, and then TLP_ENABLE set to '1' and restarted with 'systemctl restart tlp', then the CPU fans do *not* spin up.

linrunner, 2022-07-17:
Hi,

i'd try  `RUNTIME_PM_ON_AC="auto"`. 

Btw: please apply new settings with `tlp start`, not with a service restart.

kpfleming, 2022-07-17:
Thanks for the quick response! `RUNTIME_PM_ON_AC="auto"` appears to have cured the problem.

closed 2022-07-17 by kpfleming