linrunner/TLP #644: TLP does not switch cpu governor to performance

Savanta, 2022-08-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

TLP does not switch cpu governor to performance when switching from battery to AC but sets powersave when power supply is connected
When governor is set to performance via cpupower on battery and then power supply is connected governor is set to powersave

**Expected behavior**

Governor set to performance on AC

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
    Problem occur on AC
3. Actions to reproduce the behaviour
    a) Install tlp
    b) Unplug power supply
    c) Plug in power sypply
5. Shell commands entered and their output
7. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

[TLP config](https://gist.github.com/Savanta/225a5d80c987af7094044b1b35a10f41)

linrunner, 2022-08-03:
Hi, 

your data says that you're using TLP's default configuration which doesn't touch the governor at all. You must first change and activate the setting, see: https://linrunner.de/tlp/settings/processor.html. Remember to remove the leading '#'.

Savanta, 2022-08-03:
Hi,
I changed this options and still problem persists. Here's [log](https://gist.github.com/Savanta/c1b3f8e9505bbcb0a963e592507d6b2e)

linrunner, 2022-08-05:
My holiday notebook happens to be an X220 with identical processor, also running kernel 5.18. I can't reproduce your problem here, the governor changes flawlessly. 

I suggest you update your BIOS to the latest version (1.52) and reset the BIOS to default settings.

Savanta, 2022-08-05:
I'm using T420s, BIOS is custom build based on 1.52. Can the bios change governor with os booted? Also when booting governor is set properly to performance on AC and powersave on battery

linrunner, 2022-08-10:
I don't know if the BIOS can trigger a change of the governor directly, but I suspect not. The kernel (intel_pstate) gets boundary conditions like frequencies from the BIOS via ACPI. 

My crystal ball also doesn't know what's custom in your "custom build" and what settings you have chosen.

TLP has a trace mode (https://linrunner.de/tlp/support/index.html#trace-mode). In your case 

    TLP_DEBUG="pm ps rf run"

seems sufficient.

Btw: are you really shure you want the _powersave_ governor? On your hardware _intel_pstate_ runs in passive mode where the governor characteristics is the same as for  _acpi_cpufreq_. Which means that _powersave_ limits the CPU to its minimum frequency. I suggest you use _schedutil_ oder _ondemand_ instead which won't slow your laptop to a crawl and at the same time use less power.
https://linrunner.de/tlp/settings/processor.html

Savanta, 2022-08-24:
> My crystal ball also doesn't know what's custom in your "custom build" and what settings you have chosen.

I flashed [this one](https://www.bios-mods.com/forum/Thread-T420s-1-41-whitelist-removed-other-changes)
But I also flashed official with default options and problem persisted.

> Btw: are you really shure you want the powersave governor? On your hardware intel_pstate runs in passive mode where the governor characteristics is the same as for acpi_cpufreq. Which means that [...]

I don't I changed that to ondemand

Here's [output of tlp-stat -T](https://gist.github.com/Savanta/c988ded585db5680548081a15c74b693)

linrunner, 2022-08-24:
> I don't I changed that to ondemand

But your trace shows that you actually changed the config to 
```
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=ondemand
```
It also shows that TLP writes the governor sysfiles without error from the kernel (_ec=0_):

_Aug 24 14:20:14 thinkpad tlp[1462]: get_sys_power_supply(AC).ac_offline: syspwr=1
Aug 24 14:20:14 thinkpad tlp[1462]: power_source=bat
Aug 24 14:20:14 thinkpad tlp[1462]: set_cpu_scaling_governor(1): ondemand; ec=0_

_Aug 24 14:21:22 thinkpad tlp[2632]: get_sys_power_supply(AC).ac_online: syspwr=0
Aug 24 14:21:22 thinkpad tlp[2632]: power_source=ac
Aug 24 14:21:23 thinkpad tlp[2632]: set_cpu_scaling_governor(0): performance; ec=0_

I have no idea what happens after TLP properly changes the governor. Maybe the kernel silently ignores what is written.

Savanta, 2022-08-24:
> > I don't I changed that to ondemand
> 
> But your trace shows that you actually changed the config to
> 
> ```
> CPU_SCALING_GOVERNOR_ON_AC=performance
> CPU_SCALING_GOVERNOR_ON_BAT=ondemand
> ```

That's why commas matter :) "I don't want powersave, I changed that to ondemand"

> It also shows that TLP writes the governor sysfiles without error from the kernel (_ec=0_):
>
> _Aug 24 14:20:14 thinkpad tlp[1462]: get_sys_power_supply(AC).ac_offline: syspwr=1 Aug 24 14:20:14 [...]

So I did some additional testing and looks like even when tlp is not running something change governor to powersave every time power source changes. I have no idea what the hell is going on.

linrunner, 2022-08-24:
When you have solved the puzzle please let me know.

Savanta, 2022-08-31:
So I found that cat is executed on
`/sys/devices/system/cpu/cpufreq/policy0/scaling_governor`
But I have no idea from where it is executed.
This is all I have found for now
```
time->Wed Aug 31 10:38:57 2022
type=PROCTITLE msg=audit(1661935137.953:1396): proctitle=636174002F7379732F646576696365732F73797374656D2F6370752F637075302F637075667265712F7363616C696E675F676F7665726E6F72
type=PATH msg=audit(1661935137.953:1396): item=0 name="/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor" inode=17405 dev=00:15 mode=0100644 ouid=0 ogid=0 rdev=00:00 nametype=NORMAL cap_fp=0 cap_fi=0 cap_fe=0 cap_fver=0 cap_frootid=0
type=CWD msg=audit(1661935137.953:1396): cwd="/"
type=SYSCALL msg=audit(1661935137.953:1396): arch=c000003e syscall=257 success=yes exit=3 a0=ffffff9c a1=7fff1f5f3e81 a2=0 a3=0 items=1 ppid=78899 pid=79065 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="cat" exe="/usr/bin/cat" key="governor"
```

linrunner, 2022-12-04:
Any news on this before I close?