linrunner/TLP #645: Does tlp write to the EC on every single boot?

a-pav, 2022-08-13:
Hi.
I've tried to find the answer to this by my own, but eventually, I couldn't be sure.

Does tlp write to the EC on every single boot?
If it does, should we be worried about any kind of wear out on the EC, or this process causing any kind of harm to the EC/battery firmware in general?

I have an ideapad laptop, I've turned on its conservation mode through the Lenovo windows app, and now that I'm using linux, it's still in effect. This clearly shows that the necessary changes after applied once, survive boots and changes of OS. But how does TLP operate in this regard?

linrunner, 2022-08-13:
Hi,

TLP:

1. Reads the actual conservation mode -> https://github.com/linrunner/TLP/blob/main/bat.d/25-lenovo#L251
2. Writes only when the configured conservation mode differs from the actual -> https://github.com/linrunner/TLP/blob/main/bat.d/25-lenovo#L266 

ps. Nevertheless I don't see your question as a legitimate use of the issue tracker which is about bugs only.

closed 2022-08-13 by linrunner

a-pav, 2022-08-13:
I know, I know.  This wasn't an strictly legitimate issue to be raised on here, but I figured probably only a developer of the project can give a proper answer.

Nonetheless, thanks a lot for your concise answer.