linrunner/TLP #646: TLP service slow to start using systemD

Areopagitics, 2022-08-31:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I set the stop-charge setting on my battery on my ASUS laptop to 60%, but it keeps on charging past 60% for about 40 seconds until the TLP service is initialized (via systemd) so that my battery is often at 62% or even 65%. I know this is free software but it really makes it feel a bit sloppy/unprofessional. In Windows it also charges for a few seconds until the first moments of the Windows boot process, but it never really affects the battery percentage, since it triggers the ACPI call early in the boot.

**Expected behavior**

Keep battery at 60% and not charge past 60%, when set at 60%.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1.This is a battery problem.
2. Set battery to 60%.
3. Charge to 60%, and then restart a few times, and you will see the battery slowly climb past 60 up to even 65%.
4. **Full [output](https://gist.github.com/Areopagitics/a55a2f44ae3954453c47c1a446ac4d8d#file-tlp-stat-log) of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

**Additional context**

Is there any way to call the ACPI to limit the battery earlier in the boot process, similar to Windows?

linrunner, 2022-08-31:
Portraying TLP as "sloppy/unprofessional" in this issue is not an objective expression of criticism. This is also an unobjective evaluation of the contributing developers. I recommend you to use another tool.

closed 2022-08-31 by linrunner

Areopagitics, 2022-09-01:
Ouch! I thought I was providing some helpful feedback.