linrunner/TLP #647: Can't activate tlp service on Ubuntu 22.04

dwarning, 2022-08-31:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I am using Ubuntu 22.04 on a HP ProBook 6440b. Because the power management was worse on any Ubuntu upgrade in the past I decided to remove power-profiles-deamon and give a try for tlp.

tlp-stat -s
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = Hewlett-Packard  HP ProBook 6440b
BIOS           = 68CDD Ver. F.50
OS Release     = Ubuntu 22.04.1 LTS
Kernel         = 5.15.0-46-generic #49-Ubuntu SMP Thu Aug 4 18:03:25 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.15.0-46-generic root=UUID=c3b22675-9278-4af1-a9dd-5245beab30e9 ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.4)
Boot mode      = BIOS (CSM, Legacy)

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 17:30:04,      2 sec(s) ago
Mode           = battery
Power source   = battery

Error: tlp.service is not enabled, power saving will not apply on boot.
>>> Invoke 'systemctl enable tlp.service' to correct this!

So its clear that service is not started:

sudo systemctl enable tlp.service
Synchronizing state of tlp.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable tlp
Failed to enable unit: Unit file /etc/systemd/system/tlp.service is masked.

$ ls -l /etc/systemd/system/tlp.service 
lrwxrwxrwx 1 root root 9 Dez 14  2021 /etc/systemd/system/tlp.service -> /dev/null

This looks worrying for me!

All forum entries said that tlp and power-profile daemons are incompatible. But I removed the last one as I said before.

Can I use tlp for HP ProBook?
How can I start the service?

linrunner, 2022-08-31:
Try to unmask first

    sudo systemctl unmask tlp.service
    sudo systemctl enable tlp.service

From where did you install the _tlp_ package: from the official Ubuntu repositories or from my PPA? 
Please show

    dpkg -l | grep tlp

dwarning, 2022-08-31:
Thanks for the quick response!
Yes - the unmask seems to solve the problem. Enabling the service has worked for me then,
I used the Ubuntu repository:
dpkg -l | grep tlp
ii  tlp                                        1.5.0-1ubuntu3                                    all          Optimize laptop battery life
ii  tlp-rdw                                    1.5.0-1ubuntu3                                    all          Radio device wizard

Is it OK to use the Ubuntu package?

Hopeful tlp solved my problem with the worse shut-down in battery operation and the wrong percentage notification after ac plugin. Have to study tlp documentation.
Thanks again.

closed 2022-08-31 by dwarning