linrunner/TLP #650: TLP battery care started too late

braindevices, 2022-09-10:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
Asus reset the battery threshold at the beginning of the boot process. Thus it starts to charge the battery at boot time. In windows, os set the threshold shortly after windows boot starting (way earlier than TLP in linux), thus it only charges about few seconds.

The tlp.service started very late: after the multi-user.target and networkmanager.service. Usually at this time it already to late for activate the battery threshold, especially when the system has luks encryption. It may take more than few minutes to reach that point. The battery usually will charge up 0.5% to 2% during this period. It then quickly accumulated to be fully charged during a week depending on how frequently you reboot your machine. I understand there are other actions need the multi-user.target or NetworkManager.service to be fully up. But I do think the battery care do not need this. I think we can separate the batter care to a different service and start it much earlier.

**Expected behavior**

tlp should write to the battery threshold as early as possible.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Take a ASUS laptop, charge the battery > 61%
2. install tlp and config the threshold to 60.
3. reboot the laptop and keep close sight on the battery charging indicator (LED)
4. we can see the LED light up after BIOS kicked in. 
5. wait to see how long it will take to turn it off.
**Additional context**

linrunner, 2022-09-15:
Hi,

I'm aware of that. Unfortunately, there are reasons for running tlp.service so late that are not so easy to get rid of.

But you can try yourself what happens when TLP runs right at the beginning of system startup. Change **tlp.service** yourself by commenting the responsible line:

    # After=multi-user.target NetworkManager.service

braindevices, 2022-09-15:
I knew there are other functionality require tlp to run after multi-user.target. But this is because we have monolithic service. Do you think if it is possible to split it into pieces? at least we can make a separated service to handle just the battery cares.

linrunner, 2022-09-16:
Many distributions do not allow services to be enabled by the package. An additional service must then be activated by the user. Many users think they know what to activate and what not. I try to avoid this.

**I would appreciate your feedback on whether my suggestion above works.**

vortigont, 2022-09-16:
I also have asus laptop, never tested this case specifically but also never noticed an increasing battery charge. Maybe due to I usually put laptop into sleep without full shutdown. And one more note - what is your HW config? I mean, even 5-7 years old laptop with SATA SSD takes about 10 seconds to reach multiuser - not that much to get a descent amount of charge. With full encryption it does not makes much sense (I use it) because unless disk is unlocked system won't be able to execute any services aside from initramfs.

linrunner, 2023-07-15:
There seems to be no general interest here. Therefore I close.

closed 2023-07-15 by linrunner