linrunner/TLP #651: Battery Care Workaround Mode

sebastianst, 2022-09-19:
**Problem:**
Many laptop battery drivers only allow for very primitive battery care settings, e.g. the `ideapad_laptop` driver for Lenovo laptops only allows for setting either no battery charging limit (100%) or a _battery conservation mode_ that limits charging to 60%. No other charging targets can be set this way.

**Solution:** 
A workaround to allow for charging limits above the lowest limit offered by any driver is to have some sort of background process, or periodic check (e.g., every 10s with a systemd timer), that leaves the battery charging target at 100% until the intended charging target is reached, at which point it sets the charging limit to the lower limit offered by the driver.

Similarly, using a driver setting's effective charging start limits, custom charging starts could also be achieved, as long as they are above the lowest charging start that can be set by the driver.

If a driver supports more than one discrete limit (e.g., 40%, 60%, 80%), this could also be taken into account. 

Of course, this could be implemented as a standalone tool, but having this integrated into TLP would probably be less error prone, avoid having this tool and tlp get in each other's ways, and allow for a better user experience. This workaround mode would also be configurable with `tlp.conf`.

Has this, or a similar solution, already been discussed?

**Alternatives:**
Improving the driver may be another possibility in some cases. However, many battery firmwares don't even support a more fine-grained control of battery charging limits.

linrunner, 2022-09-20:
Hi, 

thanks for your interesting suggestion. Unfortunately, the approach has one major weakness:

As soon as the laptop is switched off or in suspend, the periodic monitoring does not work anymore. This worsens the user experience considerably. For the same reason, no one has (presumably) tried this with a kernel driver yet. 

There is a reason why vendor implement this in the embedded controller,  so that it works independently of the operating system.

If you absolutely need continuous limits, you have to use a corresponding laptop.

sebastianst, 2022-09-20:
Yes I anticipated this problem, it's just a workaround after all. I thought a pragmatic solution would be to just have a hook that when the system goes into suspend or powers off, TLP switches back to the lower limit of the batter firmware, so that it stop loading when the laptop is sleeping or powered off, possibly charging over the limit.

linrunner, 2022-09-20:
But this means that in this case the charging process is stopped before reaching the threshold. Your "workaround" means too much stranglehold ("Gewürge") - no offense meant. Btw: TLP has no daemon today and I will not add one for this feature.

Sorry, I will not pursue your suggestion any further.

closed 2022-09-20 by linrunner