linrunner/TLP #652: UBUNTU RELEASE: I think TLP_PERSISTENT_DEFAULT=1 should be TLP_PERSISTENT_DEFAULT=0 in ubuntu 50-estar-default.conf

gutow, 2022-10-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below) 

**Describe the bug**
On this system battery mode seems to be permanently persistent because of the setting in 50-estar-default.conf
System         = Dell Inc.  XPS 13 7390
BIOS           = 1.11.0
OS Release     = Ubuntu 22.04.1 LTS
Kernel         = 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64
Init system    = systemd v249 (249.11-0ubuntu3.6)
Boot mode      = UEFI

**Expected behavior**

When plugged into AC it should switch to AC mode.

I noticed this because I wanted to set power on under AC to my external speakers.

**To Reproduce**

On the machine and OS listed above:
1. Running on the battery issue: `sudo tlp-stat -s`. This will yield:
```
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = Dell Inc.  XPS 13 7390
BIOS           = 1.11.0
OS Release     = Ubuntu 22.04.1 LTS
Kernel         = 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.15.0-48-generic root=UUID=XXX ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.6)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 02:59:38 PM,     14 sec(s) ago
Mode           = battery (persistent)
Power source   = battery
```
2. Running plugged into power issue: `sudo tlp-stat -s`. This will yield:
```
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = Dell Inc.  XPS 13 7390
BIOS           = 1.11.0
OS Release     = Ubuntu 22.04.1 LTS
Kernel         = 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.15.0-48-generic root=UUID=XXX b ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.6)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 03:13:18 PM,     16 sec(s) ago
Mode           = battery (persistent)
Power source   = AC
```
Note that the mode insists on being set to " battery (persistent)". This is using the default settings of tlp.

**Additional context**

I can override the battery setting by issuing the command `sudo tlp ac`. But it will default back.

If I change TLP_PERSISTENT_DEFAULT=1 to TLP_PERSISTENT_DEFAULT=0 in the 50-estar-default.conf file then things work as expected. Not sure of the impact on other systems.

linrunner, 2022-10-05:
Hi, I can understand your issue. However, this is the wrong place.

Neither is the file **50-estar-default.conf** provided by TLP, nor is it part of the official Ubuntu package. 

In fact, it's part of Dell's factory image. The setting was made for the purpose of Energy Star certification. I guess the file is installed by one of the packages [here](http://dell.archive.canonical.com/updates/pool/public/o/oem-fix-misc-cnl-tlp-estar-conf/).

If you don't like it, delete it.

closed 2022-10-05 by linrunner

linrunner, 2022-10-06:
I didn't have to trace, I already _knew_ this from discussions about the **tlp.d/** scheme with a guy who did development for Dell. Long story.

https://github.com/linrunner/TLP/issues/432

You can easily query this yourself:

    dpkg -S /etc/tlp.d/50-estar-default.conf