linrunner/TLP #653: TLP do not switch from AC mode to battery

anayrat, 2022-10-05:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When laptop is unpluged from AC, tlp do not switch from AC mode to battery. Even if it detects power source is on battery.
I have to run "tlp start" in order to switch mode to battery.

**Expected behavior**

TLP should switch to battery when it detect power source on battery

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?

Both, When I plug AC, tlp stay on battery

3. Actions to reproduce the behaviour
4. Shell commands entered and their output
5. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
https://gist.github.com/anayrat/0aeef0aa6a9b4c18a390a16f47dd062f

sebastian-de, 2022-10-05:
> System         = LENOVO ThinkPad P14s Gen 2a 21A0CTO1WW
> BIOS           = R1MET49W (1.19 )

This is a firmware bug, see here: https://forums.lenovo.com/t5/Other-Linux-Discussions/Firmware-regression-No-more-udev-power-supply-events/m-p/5166407?page=1#5735942

A downgrade to v1.17 fixed it for me.

anayrat, 2022-10-05:
Wow, great, thanks!