linrunner/TLP #654: Battery control for ThinkPad X131e

niezywa, 2022-10-07:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md) [ ] I've provided all required tlp-stat outputs via [Gist](https://gist.github.com/) (see below)

On my X131e tlp-stat uses plugin "generic" for battery data. My thinkpad_acpi module is active. As far as I can understand Perl, the regex in [bat.d/05-thinkpad](https://github.com/linrunner/TLP/blob/main/bat.d/05-thinkpad) isn't going to detect this laptop.

I would appreciate if you could support my laptop. I don't know if it isn't similar enough to the x121e that I can safely edit the file.

The problem occurs on both battery and power supply

I'm sorry, but i don't want to provide my full output, as it might contain sensitive information and I don't think most of it is relevant here. Let me know if you need any specific output.

linrunner, 2022-10-07:
Hi, you already pointed out the problem, the faulty DMI product string:

> LENOVO *INVALID* 3367A17

For a single example of a 10 year old series with an incorrect string in the firmware, I will certainly not include an exception in the code.

I suggest you help yourself and patch the code for your purposes and see if it works.

linrunner, 2022-10-26:
Reading my own code, I notice that patching is not necessary. Instead you may configure the following: 

    X_SIMULATE_MODEL="ThinkPad X131e"

closed 2022-10-26 by linrunner