linrunner/TLP #656: nvidia & hibernation

wastemans, 2022-10-20:
Hello,

I have a situation where hibernation with a Lenovo P72 and nvidia xorg driver in use will not hibernate, unless:

- I'm on AC, or
- I issue the "tlp false" command

Is there any way I can get this idiot graphics card to stop getting in the way of hibernation?

At the moment I'm just using a bash script that needs to be run as root (or the sudo ticket expires):

```
$ /bin/cat /bin/ZZZ
tlp false
echo shutdown > /sys/power/disk
echo disk > /sys/power/state
sleep 1m && tlp bat
```

Thank you so much for TLP!

```
$ inxi -S
System:
  Host: XXXXXX Kernel: 5.19.15_1 arch: x86_64 bits: 64 Desktop: Fluxbox v: 1.3.7
    Distro: Void Linux
```

```
sudo tlp-stat
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
/etc/tlp.conf L0264: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0265: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
/etc/tlp.conf L0276: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0277: SOUND_POWER_SAVE_ON_BAT="1"
/etc/tlp.conf L0283: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
[... 186 lines cut ...]
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV497
/sys/class/power_supply/BAT0/cycle_count                    =    567
/sys/class/power_supply/BAT0/energy_full_design             =  99000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  90010 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  80690 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  28463 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   89.6 [%]
Capacity                                                    =   90.9 [%]

+++ Recommendations
* Install smartmontools for disk drive health info
```

linrunner, 2022-10-21:
Hi. 

in an earlier version TLP actually contained your workaround, i.e. applying AC settings before a suspend/hibernate. 

You can reactivate this old behavior by configuring:

    X_TLP_SUSPEND_ACMODE=1

https://github.com/linrunner/TLP/blob/main/tlp.in#L416

wastemans, 2022-10-21:
Thank you for the tip!

I threw that parameter into tlp and ran a few tests for fairness.

After it didn't work, I tried changing from 1 to 0, but it still won't hibernate on battery unless I use the script method.

My unenlightened guess is that the tlp service is too deep in the process tree, and therefore too late, to apply any settings that would affect hibernation once the process begins.
This might explain why my script works, but an in-process parameter does not.

wastemans, 2022-10-24:
Not a tlp bug, can't be fixed by tlp.

linrunner, 2022-10-26:
>  I threw that parameter into tlp and ran a few tests for fairness. After it didn't work

Did you test by executing `systemctl hibernate`? The script in your initial post does not invoke systemd's hooks so TLP won't be invoked as well.

> I tried changing from 1 to 0

Using 0 is pointless. 1 is the only value that brings the desired behaviour.

wastemans, 2022-10-27:
> Did you test by executing `systemctl hibernate`? The script in your initial post does not invoke systemd's hooks so TLP won't be invoked as well.

No, Void Linux is runit. There is no systemctl because there is no systemd.
You would invoke hibernation using /bin/ZZZ
Which, once all the logic is reduced down, does the same thing as my script by interacting directly with the kernel.

linrunner, 2022-10-27:
Well. A quick look at [zzz](https://github.com/void-linux/void-runit/blob/master/zzz) reveals that it provides hooks in **/etc/zzz.d/suspend/** and **/etc/zzz.d/resume/**. I don't know if the Void maintainer provisoned the necessary hooks in the _tlp_ package. See [Architecture](https://linrunner.de/tlp/developers/architecture.html#system-suspend-resume-systemd) for requirements.

linrunner, 2022-10-27:
> I don't know if the Void maintainer provisoned the necessary hooks in the _tlp_ package

As far as I can see, he did not do that: https://github.com/void-linux/void-packages/blob/master/srcpkgs/tlp/template . Only TLP's script for elogind gets installed. I think you should resolve this with an issue against the Void _tlp_ package.

wastemans, 2022-10-27:
OK, this is becoming a little beyond my comprehension here, but do I understand correctly - if I were to trigger hibernation somehow using elogind, the flags we tested might work?

linrunner, 2022-10-27:
Yes. But you can just as well create hook scripts for **zzz**:

**/etc/zzz.d/suspend/tlp**
```
#!/bin/sh
/usr/bin/tlp suspend
```

**/etc/zzz.d/resume/tlp**
```
#!/bin/sh
/usr/bin/tlp resume
```
Remember to make them executable.

wastemans, 2022-10-28:
As instructed:

```
$ for i in 'ls -l' 'cat' ; do $i /etc/zzz.d/*/tlp* ; done
-rwxr-xr-x 1 root root 30 Oct 28 17:18 /etc/zzz.d/resume/tlp
-rwxr-xr-x 1 root root 31 Oct 28 17:18 /etc/zzz.d/suspend/tlp
#!/bin/sh
/usr/bin/tlp resume
#!/bin/sh
/usr/bin/tlp suspend
```

Unfortunately, it didn't work, same as before, it hangs on hibernate unless AC is on or I use tlp to 'trick' the system into thinking it is.
I'm happy to keep using my script if you want to close.
I did close it before, assuming it was just my nvidia card at fault.

linrunner, 2022-10-29:
I'm aware that your real problem is the nvidia card and/or its driver. 

However, while looking for a workaround, I noticed that TLP doesn't seem to be called before and after hibernate (and suspend). This is not just about additionalpm functionality that you need. The reason might be the Void `tlp` package and I would like to narrow that down with your help before I close here.

What I need is a trace of a hibernate/resume cycle. For this you would have to add the following to your configuration

    TLP_DEBUG="pm ps run"

Then - on battery of course - hibernate by means of `zzz`, resume and show the following ouput:

    sudo tlp-stat --cdiff -T

wastemans, 2022-10-30:
> the Void `tlp` package and I would like to narrow that down with your help before I close here.

OK, no worries!

> add the following to your configuration
> ```
> TLP_DEBUG="pm ps run"
> ```

```
# grep DEBUG /etc/tlp.conf
TLP_DEBUG="pm ps run"
# sv restart tlp
ok: run: tlp: (pid 13274) 0s
# tlp-stat | grep 'Mode '
Mode           = battery
zzz -Z
```

At this point, Xorg stops, I get a cursor in TTY, and the system hangs indefinitely, does not power off.
Then I use Alt + sys-rq + R S E I U
Then cold power off with a long push of the power button.

> resume
Does not suspend...so I can't do exactly this.

> and show the following ouput:
> ```
> sudo tlp-stat --cdiff -T
> ```

```
# tlp-stat --cdiff -T
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0276: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0320: RUNTIME_PM_DRIVER_DENYLIST="nvidia nouveau"
/etc/tlp.conf L0032: TLP_DEFAULT_MODE="BAT"
/etc/tlp.conf L0076: CPU_SCALING_GOVERNOR_ON_AC="performance"
/etc/tlp.conf L0077: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
/etc/tlp.conf L0084: CPU_SCALING_MIN_FREQ_ON_AC="3100000"
/etc/tlp.conf L0085: CPU_SCALING_MAX_FREQ_ON_AC="4100000"
/etc/tlp.conf L0086: CPU_SCALING_MIN_FREQ_ON_BAT="800000"
/etc/tlp.conf L0087: CPU_SCALING_MAX_FREQ_ON_BAT="41000000"
/etc/tlp.conf L0481: X_TLP_SHUTDOWN_ACMODE="1"
/etc/tlp.conf L0482: X_TLP_SUSPEND_ACMODE="1"
/etc/tlp.conf L0483: TLP_DEBUG="pm ps run"

Error: /var/log/debug does not exist.

Solution: create an rsyslog conffile /etc/rsyslog.d/90-debug.conf with the following contents
 *.=debug;\
 mail,authpriv,cron.none;\
 local0,local1,local3,local4,\
 local5,local6,local7.none    -/var/log/debug
and restart the rsyslog daemon.
```

So, I did as above and re-ran the test exactly as above, and here is the debug:

```
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0276: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0320: RUNTIME_PM_DRIVER_DENYLIST="nvidia nouveau"
/etc/tlp.conf L0032: TLP_DEFAULT_MODE="BAT"
/etc/tlp.conf L0076: CPU_SCALING_GOVERNOR_ON_AC="performance"
/etc/tlp.conf L0077: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
[... 58 lines cut ...]
2022-10-30T10:24:47.374550+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.6 [0x020000]; rc=0
2022-10-30T10:24:47.378954+00:00 ur tlp[1777]: set_runtime_pm(1).deny_driver: /sys/bus/pci/devices/0000:01:00.0 [0x030000]
2022-10-30T10:24:47.383483+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:01:00.1 [0x040300]; rc=0
2022-10-30T10:24:47.387953+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:02:00.0 [0x010802]; rc=0
2022-10-30T10:24:47.392314+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:04:00.0 [0x060400]; rc=0
2022-10-30T10:24:47.396708+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:05:00.0 [0x060400]; rc=0
2022-10-30T10:24:47.401128+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:05:01.0 [0x060400]; rc=0
2022-10-30T10:24:47.405599+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:05:02.0 [0x060400]; rc=0
2022-10-30T10:24:47.410030+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:05:04.0 [0x060400]; rc=0
2022-10-30T10:24:47.414499+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:06:00.0 [0x088000]; rc=0
2022-10-30T10:24:47.418986+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:3a:00.0 [0x0c0330]; rc=0
2022-10-30T10:24:47.423533+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:70:00.0 [0xff0000]; rc=0
2022-10-30T10:24:47.427995+00:00 ur tlp[1777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:71:00.0 [0x010802]; rc=0
2022-10-30T10:24:47.525124+00:00 ur tlp[1777]: set_pcie_aspm(1).disabled_by_kernel
2022-10-30T10:24:47.527034+00:00 ur tlp[1777]: set_intel_gpu_min_max_boost_freq(1).not_configured
2022-10-30T10:24:47.533211+00:00 ur tlp[1777]: set_amdgpu_profile(1).no_gpu
2022-10-30T10:24:47.547872+00:00 ur tlp[1777]: set_wifi_power_mode(1, wlan0).iw: on; rc=0
2022-10-30T10:24:47.557658+00:00 ur tlp[1777]: disable_wake_on_lan: eth0; rc=0
2022-10-30T10:24:47.559684+00:00 ur tlp[1777]: set_sound_power_mode(1).hda: 1; rc=0
2022-10-30T10:24:47.561515+00:00 ur tlp[1777]: set_sound_power_mode(1).hda_controller: Y; rc=0
2022-10-30T10:24:47.563509+00:00 ur tlp[1777]: poweroff_drivebay(1).disabled
2022-10-30T10:24:47.617589+00:00 ur tlp[1777]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpthRGxK -> /run/tlp/run.conf
```

Hope this helps?

linrunner, 2022-11-05:
Now, the picture becomes clearer. The zzz hook calls `tlp suspend` on suspend (so also on hibernate). But AC settings are not applied despite the `X_TLP_SUSPEND_ACMODE="1"` switch. I'll see if I can reproduce this.

linrunner, 2022-11-06:
Unfortunately I cannot reproduce on my system. Please adjust the trace extent to

    TLP_DEBUG="lock pm ps run"

and perform another hibernate/resume cycle on AC. Post the output of

    sudo tlp-stat --cdiff -T -v

Since the output is quite long you may show only the relevant part (according to timestamps) or use https://gist.github.com/

wastemans, 2022-11-07:
As req'

```
2022-11-07T10:36:42.933232+00:00 ur tlp[2805]: +++ auto (1.5.0) ++++++++++++++++++++++++++++++++++++++++
2022-11-07T10:36:42.938422+00:00 ur tlp[2805]: get_sys_power_supply(AC).ac_online: syspwr=0
2022-11-07T10:36:42.939774+00:00 ur tlp[2805]: power_source=ac
2022-11-07T10:36:42.940262+00:00 ur tlp[2805]: manual_mode=ac
2022-11-07T10:36:42.940718+00:00 ur tlp[2805]: power_mode=ac
2022-11-07T10:36:42.943483+00:00 ur tlp[2805]: compare_and_save_power_state(0).equal
2022-11-07T10:36:42.961071+00:00 ur tlp[2877]: +++ resume (1.5.0) ++++++++++++++++++++++++++++++++++++++++
2022-11-07T10:36:42.966454+00:00 ur tlp[2877]: get_sys_power_supply(AC).ac_online: syspwr=0
2022-11-07T10:36:42.967659+00:00 ur tlp[2877]: power_source=ac
2022-11-07T10:36:42.968160+00:00 ur tlp[2877]: manual_mode=ac
2022-11-07T10:36:42.968674+00:00 ur tlp[2877]: power_mode=ac
2022-11-07T10:36:42.971633+00:00 ur tlp[2877]: compare_and_save_power_state(0).equal
2022-11-07T10:36:42.974684+00:00 ur tlp[2877]: set_platform_profile(0).not_configured
2022-11-07T10:36:42.976777+00:00 ur tlp[2877]: set_cpu_scaling_governor(0): performance; ec=0
2022-11-07T10:36:42.977725+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).min: 3100000; ec=0
2022-11-07T10:36:42.978305+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.978875+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.979458+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.979991+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.980551+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.981072+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.981644+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.982165+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.982693+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.983214+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.983808+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.984321+00:00 ur tlp[2877]: set_cpu_scaling_min_max_freq(0).max: 4100000; ec=0
2022-11-07T10:36:42.986950+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.987542+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu1/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.988086+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu10/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.988666+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu11/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.989201+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu2/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.989801+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu3/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.990321+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu4/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.990896+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu5/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.991525+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu6/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.992028+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu7/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.992594+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu8/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.993114+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp.write_error: balance_performance /sys/devices/system/cpu/cpu9/cpufreq/energy_performance_preference; rc=0
2022-11-07T10:36:42.993661+00:00 ur tlp[2877]: set_intel_cpu_perf_policy(0).hwp_epp: balance_performance; ec=12
2022-11-07T10:36:42.994175+00:00 ur tlp[2877]: set_intel_cpu_perf_pct(0).min.not_configured
2022-11-07T10:36:42.994720+00:00 ur tlp[2877]: set_intel_cpu_perf_pct(0).max.not_configured
2022-11-07T10:36:42.995237+00:00 ur tlp[2877]: set_cpu_boost_all(0).not_configured
2022-11-07T10:36:42.995769+00:00 ur tlp[2877]: set_intel_cpu_hwp_dyn_boost(0).not_configured
2022-11-07T10:36:42.996271+00:00 ur tlp[2877]: set_sched_powersave(0).not_available
2022-11-07T10:36:42.996887+00:00 ur tlp[2877]: set_nmi_watchdog: 0; rc=0
2022-11-07T10:36:42.998998+00:00 ur tlp[2877]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:00.0 [0x060000]; rc=0
[... cut ...]
2022-11-07T10:41:27.659462+00:00 ur tlp[8487]: +++ auto (1.5.0) ++++++++++++++++++++++++++++++++++++++++
2022-11-07T10:41:27.663128+00:00 ur tlp[8487]: get_sys_power_supply(AC).ac_online: syspwr=0
2022-11-07T10:41:27.663640+00:00 ur tlp[8487]: power_source=ac
2022-11-07T10:41:27.664188+00:00 ur tlp[8487]: manual_mode=none
2022-11-07T10:41:27.664733+00:00 ur tlp[8487]: power_mode=ac
2022-11-07T10:41:27.666340+00:00 ur tlp[8487]: lock_tlp_nb().success
2022-11-07T10:41:27.667830+00:00 ur tlp[8487]: compare_and_save_power_state(0).equal
2022-11-07T10:41:27.668788+00:00 ur tlp[8487]: unlock_tlp()

```

linrunner, 2022-11-08:
> 2022-11-07T10:39:15.437841+00:00 ur tlp[8140]: +++ suspend (1.5.0) ++++++++++++++++++++++++++++++++++++++++
> 2022-11-07T10:39:15.443629+00:00 ur tlp[8140]: get_sys_power_supply(AC).ac_online: [...]

Still doesn't contain the info I need ... I have to think about that.

wastemans, 2022-11-08:
Perhaps I need a clean power off and bootup first.
I'll gather the log again tomorrow.

linrunner, 2022-11-08:
Unnecessary. Save yourself the time.

linrunner, 2022-11-09:
Please show **via Gist**  https://gist.github.com/ the output of:

    sudo sh -x /usr/bin/tlp suspend

wastemans, 2022-11-09:
[Gist](https://gist.github.com/wastemans/e6fe729c950b246bc2cc064bd39d0983)

linrunner, 2022-11-10:
Are you sure that this debug output is complete? It breaks inside the function **save_device_states()** and never reaches the call to **apply_common_settings()** - see https://github.com/linrunner/TLP/blob/1.5.0/tlp.in#L413. This fits exactly with the symptom that no settings are applied.

Which shell is used by Void Linux for scripts?

wastemans, 2022-11-10:
This is the entire output.
Nothing indeed happens once it exits.

I'm using bash for my shell, but I believe the default is dash:
```
ls -l /bin/sh
lrwxrwxrwx 1 root root 4  1:29:06PM 25-09-2022 /bin/sh -> dash
```

wastemans, 2022-11-10:
Perhaps I could install a different shell and symlink sh to that instead?
Or just symlink sh to bash instead of dash?

linrunner, 2022-11-10:
It is sufficient if you do

    sudo bash -x /usr/bin/tlp suspend

wastemans, 2022-11-10:
No worries, I'll get you a result tomorrow.
It's not the end of the world if void is the only complication here.
We know when we stray from the norm the path isn't paved.

wastemans, 2022-11-14:
[Here you are, but it looks similar?](https://gist.github.com/wastemans/e4f43fdd7584aafbf4675da51b6b7206)

linrunner, 2022-11-17:
Not quite. The bash does what it should, settings are applied. Could you provide the relevant section of the trace after the command, e.g.

    sudo sh -x /usr/bin/tlp suspend
    sudo tlp-stat -T

wastemans, 2022-11-18:
https://gist.github.com/wastemans/d1490f875cbf7c0c9d0f421cafad74a9

linrunner, 2022-11-21:
Thanks. 

Dash terminates execution here: https://github.com/linrunner/TLP/blob/fe91d130c00d2249bb4b98d6a870db4b23e33bcb/func.d/30-tlp-func-rf-sw#L342 

Bash however continues with rc=1: https://github.com/linrunner/TLP/blob/fe91d130c00d2249bb4b98d6a870db4b23e33bcb/func.d/30-tlp-func-rf-sw#L350

The whole thing is about saving the radio devices' current state in `$RFSTATEFILE` = **/var/lib/tlp/rfkill_saved** before suspend -  which fails. Bash catches the error and continues, dash breaks. Directory **/var/lib/tlp** must be created by the package install. Does it exist on your system?

wastemans, 2022-11-22:
No :-)

```
$ ls /var/lib/tlp
ls: cannot access '/var/lib/tlp': No such file or directory
$
```

Would you like me to create this directory?

linrunner, 2022-11-22:
Indeed. Afterwards please perform yet another hibernate/resume cycle on AC. Post the output of

    sudo tlp-stat --cdiff -T

Just to be clear: the abort is a bug in dash, the missing directory is an issue with the Void tlp package.

wastemans, 2022-11-23:
My laptop now hibernates on battery without the need for my script, so you've found and fixed the bug for the void team.

Thank you, and well done!

Here is the gist as requested: [GIST](https://gist.github.com/wastemans/5913ec97590e6ae3605ecb6a16c893d3)

Happy to do more tests for you if need be.

linrunner, 2022-11-24:
Great. Thank you for your patience. 

Do you mind opening a package bug on Void about the missing directory? I would provide additional info if necessary, but I can not test a package correction here. That would be your turn.

I would then take care of a bug report for dash.

wastemans, 2022-11-24:
Done: [here](https://github.com/void-linux/void-packages/issues/40739).
Thanks linrunner, and thanks again for tlp :)

closed 2022-11-24 by wastemans

linrunner, 2022-12-02:
This does not seem to be a bug but an expected behavior of the dash. It actually complies with [POSIX defaults](https://pubs.opengroup.org/onlinepubs/9699919799/utilities/V3_chap02.html#tag_18_08_01). 

> **2.8.1 Consequences of Shell Errors**
> Redirection error with special built-in utilities: shall exit
> [...]
> Redirection error with other utilities (not special built-ins): shall not exit

The failling null command (`:`) is a "special built-in.