linrunner/TLP #658: [START/STOP_CHARGE_THRESH_BATx] unspecified behaviour, and a proposal to manage it better

paolobenve, 2022-10-31:
The docs say:

> Charging starts when connecting the charger, but only if the battery charge level is below the start charge threshold (START_CHARGE_THRESH_BATx); it will not charge when the level is above the start charge threshold
Charging stops when reaching the stop charge threshold (STOP_CHARGE_TRESH_BATx)

It says what happens when connecting the charger, and when the charging stops.

It doesn't say anything about what happens after charging stops because it has reached the top threshold. Apparently what happens is that the charge is mantained at the STOP_CHARGE_TRESH_BATx value.

However, a better (= more consistent) behaviour would be:

- charging stops at top threshold, permitting battery charge to go down
- charging starts again at bottom threshold

This behaviour would permit a continous use of the laptop with less stress on battery.

linrunner, 2022-10-31:
@paolobenve : there may be a misunderstanding on your side:
1. The behavior is hard-coded into the laptop's firmware and cannot be changed by TLP - except setting the thresholds of course
2. Discharging is not related to the charge thresholds in any way; it starts whenever you disconnect the charger -> https://linrunner.de/tlp/faq/battery.html#why-does-the-battery-not-begin-to-discharge-when-the-stop-threshold-is-reached

paolobenve, 2022-10-31:
ok, thank you for the explication.