linrunner/TLP #660: How to use the " -- CONFIG_PARAM=value" option?

Ottaviocr, 2022-11-25:
**Is your feature request related to a problem? Please describe.**

The  `-- CONFIG_PARAM=value` option  mentioned in the [man page](https://github.com/linrunner/TLP/blob/main/man/tlp.8#L93) is not properly documented or is not implemented.

**A clear and concise description of what the problem is.**

I tried multiple combinations such as :

```
sudo tlp -- TLP_ENABLE=1
sudo tlp --TLP_ENABLE=1
sudo tlp -- TLP_DEFAULT_MODE=AC
sudo tlp --TLP_DEFAULT_MODE=AC
sudo tlp -- TLP_WARN_LEVEL="3"
sudo tlp -- TLP_WARN_LEVEL=3
sudo tlp -- "TLP_WARN_LEVEL=3"
sudo tlp --"TLP_WARN_LEVEL=3"
sudo tlp "TLP_WARN_LEVEL=3"
sudo tlp TLP_WARN_LEVEL=3
sudo tlp -- TLP_WARN_LEVEL=3
```

They all return:

`Usage: tlp start|true|bat|false|ac|usb|bayoff|chargeonce|discharge|setcharge|fullcharge|recalibrate|diskid`

**Describe the solution you'd like**

Either provide examples or remove the snippet from the manual page.

linrunner, 2022-11-26:
Hi, 

keep in mind that it's an option. You must specify a complete, valid command in front of it. 

I have implemented it for the purpose of test automation only. See examples here: https://github.com/linrunner/TLP/tree/main/unit-tests. The feature is not intended for end users and there is no support for it.

But I will gladly add a note to the manpage.

Ottaviocr, 2022-11-28:
Thanks. I missed the "start" before the "--":

```
$ sudo tlp start -- LP_DEFAULT_MODE=AC
TLP started in AC mode (auto).
```

linrunner, 2022-11-28:
manpages updated.

linrunner, 2023-07-22:
Hi @Ottaviocr  : TLP 1.6 Beta 1 is out and contains the change -> https://github.com/linrunner/TLP/issues/700

Ottaviocr, 2023-07-23:
On Sat, 22 Jul 2023 at 16:23, linrunner ***@***.***> wrote:
>
> Hi @Ottaviocr : TLP 1.6 Beta 1 is out and contains the change -> #700
>

Great. Thanks.