linrunner/TLP #662: Coexistance with lg_laptop

mnlipp, 2022-11-28:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

The kernel includes a module (lg_laptop)[https://www.kernel.org/doc/html/latest/admin-guide/laptops/lg-laptop.html] which provides a facility to set the battery charge limit on lg_laptops. This kernel module is loaded automatically (no idea what triggers this) on my LG laptop (running Arch Linux). However, it is loaded after systemd has started ltp.

Loading the module sets the battery charge limit to 100%. As the module is loaded after tlp has strated, this effectively undoes tlp's setting (I have configures tlp to set the limit to 80%).

**Expected behavior**

I'm not sure how tlp sets the battery charge limit without using lg_laptop. But the startup should be delayed until the module has been loaded.

**Additional context**

A workaround is to explicitly load lg_module with a file in /etc/modules-load.d. The files there seem to be processed before systemd reached the multi-user.target.

linrunner, 2022-11-28:
tlp.service is already run really, _really_ late in the boot process. Additionally waiting for _lg_laptop_ to load at some point - trigged by  whatever - is something I want to avoid.

mnlipp, 2022-11-29:
> To be clear: TLP sets the charge thresholds via **lg_laptop**. There is no other way.

Okay, then I have misinterpreted the log. Using Arch `lg_laptop` is simply not loaded when tlp runs and therefore tlp does nothing on startup. (Yet later I see tlp reporting the wrong state because something **has** loaded `lg_laptop`. If there had been no information, I would have considered that the module is missing.) Adding `lg_laptop` to `/etc/modules-load.d` fixes this. 

My interpretation was biased because I had Fedora running on the laptop before and although there is no entry in /etc/modules-load.d, lg_laptop **is** loaded before tlp. Made me assume that it's tlp that loads the module.

closed 2022-11-29 by mnlipp

linrunner, 2022-11-29:
TLP recognizes hardware by the fact that the specific module is loaded. Then it checks if sysfiles for charge thresholds are present.

Therefore TLP presupposes that the appropriate kernel module is already loaded. The OS/kernel usually loads all modules that match the hardware. If not, the user has to enforce via configuration.

TLP does not try to load all supported kernel modules for battery care. This would clutter the syslog with failures which is not acceptable.