linrunner/TLP #664: Not discharging after reaching upper threshold

yarinbar, 2022-12-11:
Battery is fixed on the upper threshold ("STOP_CHARGE_THRESH_BAT0") and does not discharge down to "START_CHARGE_THRESH_BAT0".

In this image we can see that it is not charging but somehow it keeps the battery at the same level, might be charging in short intervals once in a few seconds to keep it steady at this level.

But i think the expected behaviour is discharging down to the START_CHARGE_THRESH_BAT0. If i pulg it in when the battery is between the two thresholds it charges it up to STOP_CHARGE_THRESH_BAT0.

[image]

`sudo tlp discharge 
Error: battery discharge/recalibrate not available.
`

# sudo tlp-stat  
`
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0029: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0053: NATACPI_ENABLE="1"
defaults.conf L0054: TPACPI_ENABLE="1"
defaults.conf L0055: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0501: START_CHARGE_THRESH_BAT0="50"
[... 173 lines cut ...]
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Care
Plugin: asus
Supported features: charge threshold
Driver usage:
* natacpi (asus_wmi) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0/1: 0(off)..100(default)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = UX425
/sys/class/power_supply/BAT0/cycle_count                    =    223
/sys/class/power_supply/BAT0/charge_full_design             =   4220 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   3354 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2649 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]

Charge                                                      =   79.0 [%]
Capacity                                                    =   79.5 [%]
`

linrunner, 2022-12-11:
You are making a mistake in your reasoning. There is a section on this in the [FAQ](https://linrunner.de/tlp/faq/battery.html#why-does-the-battery-not-begin-to-discharge-when-the-stop-threshold-is-reached).

yarinbar, 2022-12-11:
So what is the purpose of START_CHARGE_THRESH_BAT0?

Do you have an idea why `tlp discharge ` doesnt work?

linrunner, 2022-12-12:
You have to cope with what your hardware can do. ASUS laptops do not have a start charge threshold and do not have `tlp discharge`. Exactly what your output shows in the **+++ Battery Care** section and what TLP's documentation [tells](https://linrunner.de/tlp/settings/bc-vendors.html#asus). 

With your laptop you can only discharge by pulling the plug.

Please ask further comprehension questions in the Linux forum of your choice. The issue tracker is reserved for bug reports. Thank you.

closed 2022-12-15 by linrunner