linrunner/TLP #665: apple silicon (m1, m2) support

dkwo, 2022-12-13:
Could this great tool run on Apple Silicon (aarch64)?
It seems they already support things like `echo inhibit-charge > /sys/class/power_supply/macsmc-battery/charge_behaviour`

cc @marcan

linrunner, 2022-12-13:
Hi. 

From your sparse information I can't tell if hardware and drivers have the necessary capabilities, especially for charge thresholds. 

If you have the hardware, you're welcome to contribute a pull request for a suitable battery care plugin.

- [Plugin template](https://github.com/linrunner/TLP/blob/main/bat.d/TEMPLATE)
- [Kernel interface](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/ABI/testing/sysfs-class-power?id=09a0bd07764359650d41dbcb723f81321e523217#n151)
- [Required data](https://linrunner.de/tlp/settings/bc-vendors.html)

linrunner, 2022-12-15:
@SirLouen 
1. Thank you for your kind offer. I understand from your questions that you do not have the required skills. Since I do not see myself in a position to impart the necessary extensive knowledge via remote tutorial, I can unfortunately not respond to your request.
2. This not about old Macbooks, please stay on topic. Thanks.

SirLouen, 2022-12-15:
> 2. This not about old Macbooks, please stay on topic. Thanks.

Ok, I'm going to open a separate issue and work for there. I think we can reach some point and I can get there at some point if you can give me some insights on my findings.

dkwo, 2023-05-15:
happy to see a pr that implements this, closing.

closed 2023-05-15 by dkwo