linrunner/TLP #666: Intel Macbooks support for thresholds

SirLouen, 2022-12-15:
Currently, Intel Macbooks are not fully supported by TLP, specially thresholds which are the main feature i'm interested in.

I've been looking around and found this project, that somehow integrates this with TLP
https://github.com/c---/applesmc-next

Here is the template he has used:
https://github.com/c---/applesmc-next/blob/main/45-apple

I'm just commenting on this because through this, I can do some testing and you could consider integrating at some point.

SirLouen, 2022-12-15:
Looks promising

'''
# tlp-stat -b
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: APPLESMC
Supported features: charge thresholds
Driver usage:
* natacpi (APPLESMC_MOD) = active (charge thresholds)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = XINGLT
/sys/class/power_supply/BAT0/model_name                     = A1405
/sys/class/power_supply/BAT0/cycle_count                    =      5
/sys/class/power_supply/BAT0/charge_full_design             =   8000 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   8179 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   5446 [mAh]
/sys/class/power_supply/BAT0/current_now                    =   3938 [mA]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_control_full_threshold  =     80 [%]

Charge                                                      =   66.6 [%]
Capacity                                                    =  102.2 [%]
'''

Currently charging for first test.

SirLouen, 2022-12-15:
It works

```
# tlp-stat -b
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: APPLESMC
Supported features: charge thresholds
Driver usage:
* natacpi (APPLESMC_MOD) = active (charge thresholds)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = XINGLT
/sys/class/power_supply/BAT0/model_name                     = A1405
/sys/class/power_supply/BAT0/cycle_count                    =      5
/sys/class/power_supply/BAT0/charge_full_design             =   8000 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   8179 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   6523 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_control_full_threshold  =     80 [%]

Charge                                                      =   79.8 [%]
Capacity                                                    =  102.2 [%]
```

Amazing job of **c---**

linrunner, 2023-01-09:
The whole thing depends on c--- submitting his kernel patches and getting them merged. Have you contacted him to see if he will?

SirLouen, 2023-02-02:
> The whole thing depends on c--- submitting his kernel patches for the mainline kernel and getting them merged. Have you contacted him to see if he will?

Have not contacted him, but the patch is still not complete. Only thresholds are working

linrunner, 2023-02-03:
> Only thresholds are working

So what? This applies to all laptops supported so far. Only ThinkPads provide force-discharge.