linrunner/TLP #668: DEVICES_TO_(EN|DIS)_ABLE_ON_LAN_CONTROLLER

real-or-random, 2023-01-02:
**Is your feature request related to a problem? Please describe.**

My dock is not supported by tlp-rdw, and it seems that only allowlisted docks are recognized. (My dock is a Thunderbolt dock but that's not really relevant for this request.)

**Describe the solution you'd like**

Maybe hotplugging an Ethernet controller is what tlp-rdw should look for. This would probably cover 99% of the use cases why people would disable/enable for example wifi: The interesting part of plugging a dock is not the dock itself but the fact that an *Ethernet controller becomes present*. 

**Describe alternatives you've considered**

We have `DEVICES_TO_ENABLE_ON_LAN_DISCONNECT` and friends which may actually be more useful than what is suggested here. To be clear I personally prefer `DEVICES_TO_ENABLE_ON_LAN_DISCONNECT` more useful, and I probably wouldn't use the suggested feature here.

But I assume the `DOCK` variables probably exist for a reason, so I thought I'll better drop a note here. Please feel free to close.
   

**Additional context**

This could also work for thunderbolt, e.g., like this:

```
ACTION=="add|remove", SUBSYSTEM=="net", ENV{ID_PCI_SUBCLASS_FROM_DATABASE}=="Ethernet controller", RUN+="/usr/lib/udev/tlp-rdw-udev %p usb_dock"
```

(I guess there are better ways to match the subclass than matching this string -- but that's an implementation detail.)

I imagine it could work similarly for attaching USB Ethernet controllers.

linrunner, 2023-01-02:
Thanks for your suggestion. 

Unfortunately, you can't get around the dilemma you described in #667 either. Your suggested implementation would trigger exactly the same race between enabling and disabling wifi.

Besides, treating all USB ethernet adapters as docks is too quick and dirty imho.

If you were as generous with your tlp-stat outputs as you are with your wishes, we could take a look at which USB device of the dock would be suitable to be included in the corresponding [udev rules](https://github.com/linrunner/TLP/blob/main/tlp-rdw.rules.in).

real-or-random, 2023-01-02:
> Unfortunately, you can't get around the dilemma you described in #667 either.

This is not the the goal of this request. As I said, I don't need this. I'm just trying to be helpful. 

> 
> Besides, treating all USB ethernet adapters as docks is too quick and dirty imho.

I see, so let me close this.

> If you were as generous with your tlp-stat outputs as you are with your wishes, we could take a look at which USB device of the dock would be suitable to be included in the corresponding [udev [...]

It's a PCIe device (on a Thunderbolt dock), not a USB device. But I don't need this. If I'll ever need this, I'll open a PR.

closed 2023-01-02 by real-or-random