linrunner/TLP #670: TLP does not set the battery charge thresholds on a Thinkpad T14 G3

benediktahrens, 2023-01-10:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

TLP does not set the battery charge thresholds on a Thinkpad T14 G3.

**Expected behavior**

I have set TLP up to set the battery charge thresholds to 65/75% in `/etc/tlp.conf`. However, the output of `# tlp-stat -b` shows battery charge thresholds of 0/100%, and the battery is charged according to these values (aka always when plugged in).

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Edit `tlp.conf`.
2. Reboot.
3. Check output of `# tlp-stat -b`.
4. Check charge behaviour.

- [tlp.conf](https://gist.github.com/benediktahrens/87158763f9e0b36968a4bce6dc850e13)
- [output of tlp-stat --cdiff -s -b](https://gist.github.com/benediktahrens/9c92369c1d238fef654243a9ddd84546)
- [output of tlp-stat](https://gist.github.com/benediktahrens/42fd3afd50570257ec3ec7cc7c820a29)

**Additional context**

I filed a bug report with Debian on this, see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1025902.

linrunner, 2023-01-10:
Hi. I wonder why the service is disabled (because the Debain package activates it):

> Error: tlp.service is not enabled, power saving will not apply on boot.
> Invoke 'systemctl enable tlp.service' to correct this!

This causes the charge thresholds not to be applied on boot. 

Let's activate it now by command and additionally switch on the trace.

Please add the following line to your **tlp.conf**

    TLP_DEBUG="bat ps run"

Then show the output of

    sudo tlp setcharge
    sudo tlp-stat -b
    sudo tlp-stat -T

benediktahrens, 2023-01-12:
Thanks a lot for your guidance. Activating the service fails:
```
# systemctl enable tlp.service
Synchronizing state of tlp.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable tlp
Failed to enable unit: Unit file /etc/systemd/system/tlp.service is masked.
```

benediktahrens, 2023-01-12:
Maybe https://github.com/linrunner/TLP/commit/117c1065eff220c2a71a8319f2a0ede14bbedeb5 is related? `power-profiles-daemon` was automatically installed upon upgrade to Debian unstable, and was removed when I installed `tlp`.

linrunner, 2023-01-12:
Then unmask it.

benediktahrens, 2023-01-12:
Thanks! It seems to have worked:
```
# tlp setcharge 
Setting temporary charge thresholds for BAT0:
  start =  65
  stop  =  75
```
Battery status:
```
# tlp-stat -b
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B10W51864
/sys/class/power_supply/BAT0/cycle_count                    =     16
/sys/class/power_supply/BAT0/energy_full_design             =  52500 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  52500 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  23630 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   4931 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     65 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     75 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   45.0 [%]
Capacity                                                    =  100.0 [%]
```
In the following log, the relevant part starts at time `18:44`:
```
# tlp-stat -T
--- TLP 1.5.0 --------------------------------------------

Jan 12 18:08:09 nacl tlp[1799]: Error: tlp.service is not enabled, power saving will not apply on boot.
Jan 12 18:08:09 nacl tlp[1799]: >>> Invoke 'systemctl enable tlp.service' to correct this!
Jan 12 18:08:10 nacl tlp[1912]: Error: tlp.service is not enabled, power saving will not apply on boot.
Jan 12 18:08:10 nacl tlp[1912]: >>> Invoke 'systemctl enable tlp.service' to correct this!
Jan 12 18:08:10 nacl tlp[1941]: Error: tlp.service is not enabled, power saving will not apply on boot.
Jan 12 18:08:10 nacl tlp[1941]: >>> Invoke 'systemctl enable tlp.service' to correct this!
Jan 12 18:44:03 nacl tlp[9915]: +++ setcharge (1.5.0) ++++++++++++++++++++++++++++++++++++++++
Jan 12 18:44:03 nacl tlp[9915]: get_sys_power_supply(AC).ac_offline: syspwr=1
Jan 12 18:44:03 nacl tlp[9915]: power_source=bat
Jan 12 18:44:03 nacl tlp[9915]: manual_mode=none
Jan 12 18:44:03 nacl tlp[9915]: power_mode=bat
Jan 12 18:44:03 nacl tlp[9915]: check_thinkpad: tpmodel=T14 Gen 3
Jan 12 18:44:03 nacl tlp[9915]: batdrv_init.thinkpad: batteries=BAT0; natacpi=0; tpacpi=256; tpsmapi=254
Jan 12 18:44:03 nacl tlp[9915]: batdrv_init.thinkpad: read=natacpi; thresh=natacpi; bn_start=charge_control_start_threshold; bn_stop=charge_control_end_threshold; dischg=natacpi; bn_dischg=charge_behaviour
Jan 12 18:44:03 nacl tlp[9915]: batdrv.thinkpad.select_battery(DEF): bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; bf_start=/sys/class/power_supply/BAT0/charge_control_start_threshold; bf_stop=/sys/class/power_supply/BAT0/charge_control_end_threshold; bf_dischg=/sys/class/power_supply/BAT0/charge_behaviour
Jan 12 18:44:03 nacl tlp[9915]: batdrv.thinkpad.read_threshold(start): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_start_threshold; bat_idx=1; out=0; rc=0
Jan 12 18:44:03 nacl tlp[9915]: batdrv.thinkpad.read_threshold(stop): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_end_threshold; bat_idx=1; out=100; rc=0
Jan 12 18:44:03 nacl tlp[9915]: batdrv.thinkpad.write_thresholds(65, 75, 2, BAT0).start.write: bat=BAT0; old=0; new=65; steprc=0
Jan 12 18:44:03 nacl tlp[9915]: batdrv.thinkpad.write_thresholds(65, 75, 2, BAT0).stop.write: bat=BAT0; old=100; new=75; steprc=0
Jan 12 18:44:03 nacl tlp[9915]: batdrv.thinkpad.write_thresholds(65, 75, 2, BAT0).complete: bat=BAT0; rc=0
Jan 12 18:44:03 nacl tlp[9998]: +++ auto (1.5.0) ++++++++++++++++++++++++++++++++++++++++
Jan 12 18:44:03 nacl tlp[9998]: get_sys_power_supply(AC).ac_offline: syspwr=1
Jan 12 18:44:03 nacl tlp[9998]: power_source=bat
Jan 12 18:44:03 nacl tlp[9998]: manual_mode=none
Jan 12 18:44:03 nacl tlp[9998]: power_mode=bat
Jan 12 18:44:03 nacl tlp[9998]: compare_and_save_power_state(1).equal
Jan 12 18:44:03 nacl tlp[10031]: +++ auto (1.5.0) ++++++++++++++++++++++++++++++++++++++++
Jan 12 18:44:03 nacl tlp[10031]: get_sys_power_supply(AC).ac_offline: syspwr=1
Jan 12 18:44:03 nacl tlp[10031]: power_source=bat
Jan 12 18:44:03 nacl tlp[10031]: manual_mode=none
Jan 12 18:44:03 nacl tlp[10031]: power_mode=bat
Jan 12 18:44:04 nacl tlp[10031]: compare_and_save_power_state(1).equal
```

linrunner, 2023-01-12:
The service was most likely masked by yourself, I don't know of any other Debian packages doing that. Because p-p-d blocks TLP and several other tools (https://github.com/linrunner/TLP/issues/564) it is displaced by the Debian package via "Conflicts:".

benediktahrens, 2023-01-12:
I am pretty sure I would remember masking the service; I did not know about masking before today. Unless masking happens as a side effect of another action...

benediktahrens, 2023-01-12:
Thanks a lot for guiding me; I learned a lot!

linrunner, 2023-01-12:
You're welcome. Nevertheless, a service does not mask itself alone. I cannot tell you what the cause is in your case.

MoonSweep, 2023-01-13:
So this was not a bug after all, but only a configuration problem ? None of your previous reports on the Debian BTS mentioned that the service was not started...

MoonSweep, 2023-01-13:
So this was not a bug after all, but only a configuration problem ? None of your previous reports on the Debian BTS mentioned that the service was not started... May I close the bug there, then ?

benediktahrens, 2023-01-13:
Yes, please close the bug report. I still don't know how the service got 
deactivated/masked, though.
Thanks a lot for your help!

closed 2023-01-13 by linrunner