linrunner/TLP #671: TLP / Kernel 6.2 / Gentoo

anathonous, 2023-01-14:
**Describe the bug**

scaling_max_freq gets set to 400mhz and isn't switching.

**Expected behavior**

pstate settings should change. 

**To Reproduce**

Steps to reproduce the unexpected behavior:

Does the problem occur on battery or AC or both?
Both

Actions to reproduce the behaviour.
Boot computer
t

--- TLP 1.5.0 --------------------------------------------

+++ Processor
CPU model      = Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = performance
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2900000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = performance [HWP.EPP]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu1..cpu3: omitted for clarity, use -v to show all

/sys/devices/system/cpu/intel_pstate/status            = active
/sys/devices/system/cpu/intel_pstate/min_perf_pct      =  13 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   0
/sys/devices/system/cpu/intel_pstate/hwp_dynamic_boost =   1
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  23 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  27

/sys/module/workqueue/parameters/power_efficient       = N
/proc/sys/kernel/nmi_watchdog                          = 1

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = (not available)
/sys/firmware/acpi/platform_profile_choices            = (not available)
/sys/devices/platform/thinkpad_acpi/dytc_lapmode       = (not available)
**Additional context**

It's a Lenovo X1 Carbon Gen 5 with Gen 9 Intel Skylake. 
The issue appears to be with the way pstate max_freq and minfreq are set for the scaling driver.

I can manually go in and set it.
/sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq

Any insight would be great. I am using 6.2 RC 3 on Gentoo ~amd64 Testing.
-Nate

linrunner, 2023-01-14:
Your issue cannot be processed without complete outputs as described in the [issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

Provide them via https://gist.github.com/ please.

anathonous, 2023-01-14:
sudo tlp-stat
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0029: TLP_ENABLE="0"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
/etc/tlp.conf L0116: CPU_ENERGY_PERF_POLICY_ON_AC="performance"
/etc/tlp.conf L0117: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0152: SCHED_POWERSAVE_ON_AC="0"
/etc/tlp.conf L0153: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
/etc/tlp.conf L0177: DISK_DEVICES="nvme0n1 sda"
/etc/tlp.conf L0185: DISK_APM_LEVEL_ON_AC="254 254"
/etc/tlp.conf L0186: DISK_APM_LEVEL_ON_BAT="128 128"
/etc/tlp.conf L0196: DISK_APM_CLASS_DENYLIST="usb ieee1394"
/etc/tlp.conf L0225: DISK_IOSCHED="mq-deadline mq-deadline"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
/etc/tlp.conf L0254: AHCI_RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0255: AHCI_RUNTIME_PM_ON_BAT="auto"
/etc/tlp.conf L0261: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.conf L0343: PCIE_ASPM_ON_AC="performance"
/etc/tlp.conf L0344: PCIE_ASPM_ON_BAT="powersave"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
/etc/tlp.conf L0315: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0316: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0321: WOL_DISABLE="Y"
/etc/tlp.conf L0329: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0330: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
/etc/tlp.conf L0269: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0270: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
/etc/tlp.conf L0349: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0350: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
/etc/tlp.conf L0381: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0053: NATACPI_ENABLE="1"
defaults.conf L0054: TPACPI_ENABLE="1"
defaults.conf L0055: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0087: CPU_SCALING_GOVERNOR_ON_AC="schedutil"
/etc/tlp.conf L0088: CPU_SCALING_GOVERNOR_ON_BAT="schedutil"
/etc/tlp.conf L0099: CPU_SCALING_MIN_FREQ_ON_AC="0"
/etc/tlp.conf L0100: CPU_SCALING_MAX_FREQ_ON_AC="100"
/etc/tlp.conf L0101: CPU_SCALING_MIN_FREQ_ON_BAT="0"
/etc/tlp.conf L0102: CPU_SCALING_MAX_FREQ_ON_BAT="90"
/etc/tlp.conf L0126: CPU_MIN_PERF_ON_AC="0"
/etc/tlp.conf L0127: CPU_MAX_PERF_ON_AC="100"
/etc/tlp.conf L0128: CPU_MIN_PERF_ON_BAT="0"
/etc/tlp.conf L0129: CPU_MAX_PERF_ON_BAT="75"
/etc/tlp.conf L0136: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0137: CPU_BOOST_ON_BAT="1"
/etc/tlp.conf L0145: CPU_HWP_DYN_BOOST_ON_AC="1"
/etc/tlp.conf L0146: CPU_HWP_DYN_BOOST_ON_BAT="1"
[... 184 lines cut ...]
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV431
/sys/class/power_supply/BAT0/cycle_count                    =    552
/sys/class/power_supply/BAT0/energy_full_design             =  57020 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  46620 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  44680 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   95.8 [%]
Capacity                                                    =   81.8 [%]

+++ Recommendations
* Install smartmontools for disk drive health info

nkoch@Space ~ $

linrunner, 2023-01-14:
Until you fix this, there will be no support here:

> Error: tlp.service is not enabled, power saving will not apply on boot.
>             Invoke **'systemctl enable tlp.service**' to correct this!
> 
> ***Warning: laptop-mode-tools detected, this may cause conflicts with TLP.
> **Please uninstall laptop-mode-tools.**

Afterwards I need the outputs for AC and Battery, as the template says. I would also like to point out that outputs should be provided via Gist, otherwise it will be too messy. Thanks.

anathonous, 2023-01-14:
CPU_SCALING_FREQ was the issue. They need to be commented out. I got help on reddit.

linrunner, 2023-01-14:
Just for the record: you've got quite a lot of issues with your configuration.

> /etc/tlp.conf L0087: CPU_SCALING_GOVERNOR_ON_AC="schedutil"
> /etc/tlp.conf L0088: CPU_SCALING_GOVERNOR_ON_BAT="schedutil"

The _intel_pstate_ driver does not provide a _schedutil_ scaling governor. So your setting is ignored CPU ends up with the _performance_ governor:

> /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver = intel_pstate
> /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor = **performance**
> /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = **performance powersave**

Use _powersave_ instead.

Which means no power saving and max CPU frequency all the time together with the side effect that these settings won't apply:

> /etc/tlp.conf L0116: CPU_ENERGY_PERF_POLICY_ON_AC="performance"
> /etc/tlp.conf L0117: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"

You have already noticed these invalid values, frequencies must be in kHz, otherwise the driver ignores them. 

> /etc/tlp.conf L0100: CPU_SCALING_MAX_FREQ_ON_AC="100"
> /etc/tlp.conf L0102: CPU_SCALING_MAX_FREQ_ON_BAT="90"

I recommend to either set frequencies _or_ limit performance by percentage, but not both:

> /etc/tlp.conf L0127: CPU_MAX_PERF_ON_AC="100"
> /etc/tlp.conf L0129: CPU_MAX_PERF_ON_BAT="75"

Since the performance governor is active, it doesn't really matter what you configured, the driver will ignore it anyway.

closed 2023-01-14 by linrunner

anathonous, 2023-01-14:
https://wiki.gentoo.org/wiki/User:Nathanlkoch/Tutorials/Lenovo/Hardware/X1Carbon5thGen

I'm using a conjunction of both laptop-mode and tlp. laptop-mode has some nice exec features.
I will keep tweaking.

Thank you.

linrunner, 2023-01-14:
Of course, you are free to do so. But please spare me issues about the unpredictable results of such a system configuration.