linrunner/TLP #673: Request: Improve documentation on unsupported hardware

jluttine, 2023-01-22:
**Is your feature request related to a problem? Please describe.**

When battery care is not supported, it'd be nice to know more about the reason or what can be done if anything. The documentation could be a bit more explicit about this.

I'm referring to this section:
https://linrunner.de/tlp/settings/bc-vendors.html#unsupported-hardware

**Describe the solution you'd like**

I'd like to see a bit more information in that section on what "unsupported hardware" means.

- Does it mean that there is no way to ever get the battery care features because the hardware is missing something?
- Does it mean that TLP currently doesn't support the hardware but support could be added?
- Does it mean that some other component (e.g., kernel, driver, firmware) is missing support for the hardware features but they could be added in those components?

Now, I'm a bit lost what I can do if anything. "Unsupported hardware" is a bit vague term to me.

So, could this information be added to that section?

Also, the section says that for unsupported hardware, the output of `tlp-stat -b` looks like this:

```
+++ Battery Care
Plugin: generic
Supported features: none available
```

But in my case, the output is:

```
+++ Battery Care
Plugin: asus
Supported features: none available
Driver usage:
* natacpi (asus_wmi) = inactive (laptop not supported)
```

So, they are different because Plugin is not `generic` but `asus`. So, I'm not sure if my output is what unsupported hardware output is expected to be. I suppose so, but maybe if `Plugin` can be something other than `generic`, the example output in the docs could be given as:

```
+++ Battery Care
...
Supported features: none available
...
```

instead of fixing `Plugin` to be `generic` if it doesn't need to be. Here, `...` means that there can be anything and the only thing that is relevant is the line `Supported features`.

**Describe alternatives you've considered**

-

**Additional context**

-

linrunner, 2023-01-22:
`Plugin: asus` indicates that an ASUS laptop has been detected via the loaded `asus_wmi` driver, but the driver does not provide a working stop charge threshold sysfile **/sys/class/power_supply/BAT0/charge_control_end_threshold** (https://github.com/linrunner/TLP/blob/main/bat.d/15-asus#L100). But TLP cannot find out on which level - hardware, firmware, kernel driver - the problem lies. The required kernel version is given in the TLP documentation.

I am not convinced that detailed explanations about possible causes really help. Of course, no user likes it when a feature is `not supported`. But if you really want to delve deeper, you have to have corresponding engineering knowledge anyway.

jluttine, 2023-01-22:
Ah, ok, well that's already good information to know that the problem can be on any level (hardware, firmware, kernel driver). I mean, if one can say what are the possible causes, that already helps a lot. So, if those are the levels where the issue can be, then perhaps that could be stated in the docs? Thanks anyway!

linrunner, 2023-01-22:
I have expanded the documentation. Thanks for the suggestion.

closed 2023-01-22 by linrunner