linrunner/TLP #674: emulating charge_control_start_threshold

axet, 2023-01-22:
Hello!

On systems with only 'charge_control_end_threshold' available we can emulate  charge_control_start_threshold by settings charge_control_end_threshold to the charge_control_start_threshold. As result we can properly support charge_control_start_threshold and charge_control_end_threshold logic.

Using star_threshold and end_threshold will prevent battery charging spam, and reduce charging attempts until 10% of battery charge is lost. Having only one charge_control_end_threshold option only prevent overcharing, but not recharging spam. Luckly for us, we can emulate start_threshold.

**Describe the solution you'd like**

tlp service monitor battery level. When battery level reaching charge_control_end_threshold service set charge_control_end_threshold to the charge_control_start_threshold wait until battery level charge_control_end_threshold reaching charge_control_start_threshold. Set charge_control_end_threshold to the charge_control_end_threshold.

closed 2023-01-22 by linrunner

axet, 2023-01-23:
For anyone who looking for "smart" charge control with emulation start_threshold I wrote a simple scrip which can be installed as systemd service:

* https://gitlab.com/axet/homebin/-/blob/debian/charge-control

linrunner, 2023-01-23:
Hi.

Sorry. Something must have gone wrong when saving my answer. So here we go:

In principle, you can do it the way you suggest. In practice I doubt that it works reliably, comprehensible and predictable from a user perspective. Monitoring is not possible in suspend and off state. There is a reason why charge thresholds are implemented in hardware (more precisely: embedded controller firmware).

By the way, you are not the first with this suggestion. Nevertheless, I will not implement it.

axet, 2023-01-23:
I think it is not very accurate statement. First of all, here is no persistence between reboots for threshold settings. No body have control for off-line states. Neither my script, or /sys variables, nor tlp. Second, suspend / resume cycles should relay on last value set to the end_threshold. If it have been set to START or END value it will persist until resume and theoretically will work properly (I haven't tested it).

Anyway, no problem here. Author know the best where his software need to go. Anyone who need script have it (above link).

Currently debian and ubuntu have 'power-profiles-daemon' which replaces 'tlp' and have support for charging control planned (https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1461)