linrunner/TLP #676: Asus CHARGER_END_THRESHOLD does not work

ldrahnik, 2023-02-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Setting up value in file `sudo gedit /etc/tlp.conf`:

```
#START_CHARGE_THRESH_BAT0=0 my laptop does not support it probably because file `test /sys/class/power_supply/BAT0/charge_control_start_threshold` does not exist
STOP_CHARGE_THRESH_BAT0=60
```

does not stop charging, laptop `UP5401EA`.

**Expected behavior**

Changed value in file `cat /sys/class/power_supply/BAT0/charge_control_end_threshold` should stop charging level -1 number. (`sudo tlp start` or `reboot` also did not help stop charging when was set up threshold in config)

**To Reproduce**

I can use my laptop/hardware to test it. I already installed `clitest` and looked a little in the script for battery in asus laptops.

**Additional information**

Changing file directly on my laptop works `sudo gedit /sys/class/power_supply/BAT0/charge_control_end_threshold`.

linrunner, 2023-02-02:
Your issue cannot be processed without complete outputs as described in the [issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

Provide them via https://gist.github.com/ please.

ldrahnik, 2023-02-02:
@linrunner link to `tlp-stat` - https://gist.github.com/ldrahnik/3086c5d72a5debffc8a188dbaac7883a

linrunner, 2023-02-02:
> --- TLP 1.3.1

You need TLP 1.4 or newer for that.

ldrahnik, 2023-02-02:
@linrunner works, my bad