linrunner/TLP #677: CPU freqency not going to maximum when setting default governor to performance in Intel CPUs

OshankKashyap, 2023-02-03:
I'm attempting to use the performance governor to set the CPU frequency to maximum when the power is on Battery. I'm using the following configuration: -
```
# Test File
TLP_ENABLE=1

# Battery Configuration
#START_CHARGE_THRESH_BAT0=0
#STOP_CHARGE_THRESH_BAT0=0

# CPU Configuration
CPU_MIN_PERF_ON_AC=10
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=10
CPU_MAX_PERF_ON_BAT=100

CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=1

CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=performance

CPU_ENERGY_PERF_POLICY_ON_AC=performance
CPU_ENERGY_PERF_POLICY_ON_BAT=performance
```

When using this configuration I'm getting - 
```
pstate-frequency <3.12.0>
    CPU_DRIVER   -> intel_pstate
    CPU_GOVERNOR -> performance
    TURBO        -> 0 [ON]
    CPU_MIN      -> 10% [400MHz]
    CPU_MAX      -> 47% [1833MHz]
```

The 'CPU MAX' setting should be '3900MHz,' but for some reason, it isn't. I tried changing the maximum performance, but the problem persists.

I'm not sure if it's a bug or if I'm not messing with the configuration.

linrunner, 2023-02-03:
Your issue cannot be processed without complete outputs as described in the [issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

Provide them via https://gist.github.com/ please.
_______________________________________________

OshankKashyap, 2023-02-18:
Updated! Please check it

linrunner, 2023-02-20:
I don't see an issue with TLP here, your settings are applied correctly.

I guess the problem is the _intel_pstate_ driver, which does not release the turbo frequencies. Possibly also a restriction imposed by the BIOS. At least nothing that would be in scope here.

OshankKashyap, 2023-02-23:
I checked my BIOS settings but everything seems to be good in shape. Before TLP I was using this thing [pstate-frequency](https://github.com/pyamsoft/pstate-frequency) and it was able to set the CPU frequency to maximum when on Battery.

Here is the command that I ran -
`sudo /usr/local/bin/pstate-frequency --set --plan performance`

Output:
```
pstate-frequency <3.12.0>
    CPU_DRIVER   -> intel_pstate
    CPU_GOVERNOR -> performance
    TURBO        -> 0 [ON]
    CPU_MIN      -> 10% [400MHz]
    CPU_MAX      -> 100% [3900MHz]
```

Here's the tlp stat - [https://gist.github.com/OshankKashyap/5534c23fd280f620040e1b92e3d8eebf](https://gist.github.com/OshankKashyap/5534c23fd280f620040e1b92e3d8eebf)

linrunner, 2023-02-24:
I have looked at the pstate-frequency code. Interesting.

It not only writes **/sys/devices/system/cpu/intel_pstate/{min,max}_perf_pct**, but also calculates and writes **/sys/devices/system/cpu/cpu0/cpufreq/scaling_{min.max}_freq** accordingly. 

TLP does not do the latter without additional configuration. With the systems I know so far, it is not necessary.

Try to add the following to your config

```
CPU_SCALING_MIN_FREQ_ON_AC=400000
CPU_SCALING_MAX_FREQ_ON_AC=3900000
CPU_SCALING_MIN_FREQ_ON_BAT=400000
CPU_SCALING_MAX_FREQ_ON_BAT=3900000

```

OshankKashyap, 2023-02-26:
It is working as expected.
Thanks!

closed 2023-02-28 by OshankKashyap

linrunner, 2023-03-02:
Thanks for your feedback.