linrunner/TLP #679: /sys/class/dmi/id/board_name allows identification of librebooted devices.

karstenes, 2023-02-23:
**Is your feature request related to a problem? Please describe.**

This will allow proper identification of librebooted thinkpads.

As discussed in [this issue](https://github.com/linrunner/TLP/issues/657), tlp does not identify thinkpads with coreboot due to an improper /sys/class/dmi/id/product_version string. The /sys/class/dmi/id/board_name is properly populated however, which will allow the identification.

**Describe the solution you'd like**

Utilization of the /sys/class/dmi/id/board_name string for tlp battery driver identification.

**Additional context**
Here is a gist of all the strings in /sys/class/dmi/id/*

[https://gist.github.com/karstenes/c15397843e097bc273bf507a216ef031](https://gist.github.com/karstenes/c15397843e097bc273bf507a216ef031)

linrunner, 2023-02-24:
Hi,

I confess that frequent incorporation of workarounds is tiring. I would really prefer Libreboot to fix its stuff. But so be it.

Please test and report.

karstenes, 2023-02-24:
Works great. With tp_smapi loaded the thinkpad is detected properly.

linrunner, 2023-02-24:
Hold on. No one gets out of here without outputs! Show:

`sudo tlp-stat -s -b`

karstenes, 2023-02-24:
Sorry!

Here's the tlp-stat

https://gist.github.com/karstenes/b0a04a0cd609fb980560bdb044cd6a97

linrunner, 2023-02-25:
This output does not fit yet
> +++ System Info
> System         = LENOVO **1.0** ThinkPad T440p
Stay tuned.

linrunner, 2023-02-25:
Done. Please test again and show

    sudo tlp-stat -s -b

Btw: you don't need external kernel modules with your ThinkPad model

- _tp_smapi_ is for T420 and older only
- _acpi_call_ is obsolete with kernel 5.17 or newer

Does setting the charge thresholds work with Libreboot?

As for force-discharge (recalibration), I expect from your output that with Libreboot it does not work:
> /sys/class/power_supply/BAT0/charge_behaviour               = [auto]

You don't need to try external kernel modules, the necessary logic is available from kernel 5.17 on. If it is to work, it must look like this:
> /sys/class/power_supply/BAT0/charge_behaviour               = [auto] **inhibit-charge force-discharge**

karstenes, 2023-02-26:
Here's that gist with that output.
https://gist.github.com/karstenes/e440b94d3d244b954c7c65ce71d7587f

linrunner, 2023-02-27:
Output looks fine. Thanks for testing.

closed 2023-02-27 by linrunner

linrunner, 2023-07-22:
Hi @karstenes  : TLP 1.6 Beta 1 is out and contains the change -> https://github.com/linrunner/TLP/issues/700