linrunner/TLP #681: Support end of line comments

OJFord, 2023-03-10:
**Is your feature request related to a problem? Please describe.**

I was tearing my hair out thinking my drop-in configuration wasn't being read, checking versions, trying `.conf.d` instead (despite the documented and installed `.d` location), removed to `tlp.conf` which then complained it couldn't find it, seemingly supporting that.

I eventually found this little nugget in the docs:
> Comments after parameters are not allowed, the whole line will be silently ignored:
>
> EXAMPLE="dont use like this" # Parameter in front is ignored!

Which is very counter-intuitive, and just silently failing (even though it's documented - I wouldn't think to look for it when it's so common, I know whole-line comments are supported, and it doesn't error).

**Describe the solution you'd like**

I'd like it to be allowed, it's particularly useful in `USB_BLACKLIST` or similar lists of magic numbers:
```
USB_BLACKLIST+="1337:dead" # Acme Keyboard One
USB_BLACKLIST+="1337:beef" # Acme Mouse 3000
```

**Describe alternatives you've considered**

Parse error to make the problem a lot more apparent.

**Additional context**

`tlp-stat -u` would show "(not configured)" after `systemctl restart tlp`.

linrunner, 2023-03-11:
Hi, in fact this is already implemented. Coming with Release 1.6.

https://github.com/linrunner/TLP/blob/main/changelog#L13
https://github.com/linrunner/TLP/commit/135756dbd36c6c980de51fdc1160a499e483a304

closed 2023-03-11 by linrunner

OJFord, 2023-03-11:
Oh fantastic, thanks!

linrunner, 2023-07-22:
Hi @OJFord  : TLP 1.6 Beta 1 is out and contains the change -> https://github.com/linrunner/TLP/issues/700