linrunner/TLP #682: Error: cannot write runtime configuration

nurmohammed840, 2023-03-13:
Recently I got some error...

```
❯ sudo tlp start
/usr/share/tlp/tlp-func-base: line 241: id: command not found
/usr/share/tlp/tlp-func-base: line 554: mktemp: command not found
Error: cannot write runtime configuration to .
```

I also reinstall `tlp` on my fedora.

```
> sudo dnf install tlp tlp-rdw
> systemctl enable tlp.service
> systemctl mask systemd-rfkill.service systemd-rfkill.socket
```

But this does not resolve this error...

```
❯ which mktemp & id
╭───┬────────┬─────────────────┬──────────╮
│ # │  arg   │      path       │ built-in │
├───┼────────┼─────────────────┼──────────┤
│ 0 │ mktemp │ /usr/bin/mktemp │ false    │
│ 1 │ id     │ /usr/bin/id     │ false    │
╰───┴────────┴─────────────────┴──────────╯

```

nurmohammed840, 2023-03-13:
I also checked:

```yml
❯ sudo tlp-stat -b
[sudo] password for nur: 
/usr/share/tlp/tlp-func-base: line 241: id: command not found
/usr/share/tlp/tlp-func-base: line 554: mktemp: command not found
Error: cannot write runtime configuration to .
/usr/share/tlp/tlp-func-base: line 241: id: command not found
Error: missing root privilege.
```

❯ bat /usr/share/tlp/tlp-func-base

```sh
 239   │ test_root () {
 240   │     # test root privilege -- rc: 0=root, 1=not root
 241   │     [ "$(id -u)" = "0" ]
 242   │ }
```

Is there something with root privilege ?
But I did run with `sudo`

linrunner, 2023-03-13:
Very strange indeed. Which Fedora is this exactly, the workstation? Version?

What is the output of

    tlp-stat -s
    sudo which id mktemp

nurmohammed840, 2023-03-13:
Yes, Fedora 37 (workstation) 
kernel version: 6.1

linrunner, 2023-03-14:
I would still like to see the outputs requested above.

linrunner, 2023-03-14:
OK, then I'll close here.

closed 2023-03-14 by linrunner