linrunner/TLP #683: Recalibrate shutting down when dischare complete rather than beginning full charge

john-cabaj, 2023-03-24:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Computer shuts down after `tlp recalibrate BATx` rather than starting to charge

**Expected behavior**

My understanding of `tlp recalibrate` is that it fully discharges, then switches to charging to 100% once complete.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? 
    AC power, else I can't run `tlp recalibrate`
2. Actions to reproduce the behaviour
    Run `tlp recalibrate` on Ubuntu 22.04.2, with 5.19 kernel
3. Shell commands entered and their output
    `tlp recalibrate BAT0`
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
   
   `$ sudo tlp-stat 
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
[... 168 lines cut ...]
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 5B10W51826
/sys/class/power_supply/BAT0/cycle_count                    =      9
/sys/class/power_supply/BAT0/energy_full_design             =  50500 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  50500 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  36730 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  37449 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   72.7 [%]
Capacity                                                    =  100.0 [%]

+++ Recommendations
* Install smartmontools for disk drive health info`

**Additional context**

N/A

linrunner, 2023-03-24:
First of all: TLP does not have a function that shuts down the system, so I assume it must be something else.

What exactly happens during this "shutdown"?

a. The machine powers off suddenly
b. Ubuntu shuts down (with corresponding messages on the console)

Case a. would be a hardware issue. 

Case b. could be that Ubuntu does not take the connected charger into account and thinks it has to shut down because of a low charge level. However, this would not be the normal behavior of Ubuntu. On my machines, such a thing has never occurred during recalibration.

At what charge level (percentage) does the phenomenon occur?
Does Ubuntu display any messages on the desktop before shutdown?

john-cabaj, 2023-03-24:
It was definitely case A. The charge threshold was well below 0% and continued depleting. The mV dropped to around 9000 last I looked at it. Then the machine had a hard shutdown, straight to black screen. 

I was using the stock charger with a Thinkpad P14s Gen 2 (AMD). At no point did it come off AC power.

closed 2023-03-31 by linrunner

john-cabaj, 2023-03-31:
I try not to recalibrate too often. My computer otherwise charges fine, and a “manual” recalibration (full discharge, full charge) yielded a higher “energy_full” value. So all things considered, I was able to recalibrate. I just didn’t expect the computer to shut down halfway through.

linrunner, 2023-03-31:
With the charger connected, the ThinkPad must not power off under any circumstances during recalibration. My own P14s Gen 2 AMD does not exhibit this behavior.

john-cabaj, 2023-03-31:
Could be the kernel. I’m in between 5.15 (no built-in calibration support without additional modules) and 5.19 (built-in calibration support). I’ll give it another shoot soon.

yellowjacketcoder, 2023-07-26:
@linrunner I have experienced the same issue in my L14 Gen2. It's tlp-stat is as follows:

```
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0053: NATACPI_ENABLE="1"
defaults.conf L0054: TPACPI_ENABLE="1"
defaults.conf L0055: TPSMAPI_ENABLE="1"

+++ System Info
System         = LENOVO ThinkPad L14 Gen 2 20X1S0Q700
BIOS           = R1JET62W (1.62 )
OS Release     = Ubuntu 22.04.2 LTS
Kernel         = 5.19.0-50-generic #50-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 10 18:24:29 UTC 2023 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.19.0-50-generic root=UUID=f0f3bcc6-f2eb-4008-8c5f-2e9087219fc9 ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.9)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
[... 158 lines cut ...]

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   99.9 [%]
Capacity                                                    =   15.5 [%]

+++ Recommendations
* Install smartmontools for disk drive health info
```

I never had to recalibrate my battery and because I almost always keep my laptop plugged in I had set START as 40 and STOP at 50 sacrificing runtime for battery health.

Usually, with 50% if I had to disconnect the charger(lets say for a meeting or a break) it would still run on battery for ~45 mins to an hour.

But recently( in past week) I noticed that it didn't even give me 15 mins of backup on battery at 50%.

So I thought why not simply `fullcharge` the system once? So I did that, I ran `sudo tlp fullcharge BAT0` and it charged the system to 100% and then I disconnected it from AC power, then too it gave me ~20-25 mins which is unusually low.

I tried the `sudo tlp fullcharge BAT0` again but this time I actually forgot that I had put it on fullcharge and did NOT disconnect the charger. At the end of my workday some 3-4 hours later I disconnected the charger and lo and behold, it gives me full ~4 hours on battery doing the same things I did in previous runs(mostly wrote code, some internet browsing). Which led me to believe that the actual battery is alright but there is some issue in percentages.

Tried to recalibrate, faced the same issue as @john-cabaj of laptop hard shutting down when it nears to almost depleted during the forced discharge while the ac charger was still plugged in.

Currently tlp-stat shows energy_full = 6990 but I know for a fact that that aint the case because if I remove the thresholds and charge it tlp-stat stops reporting the increase in energy after `energy_now` reaches 6980(99.9% of energy_full) but if I remove the charger right then it doesn't decrease at all for a few minutes and then starts decreasing.

linrunner, 2023-07-26:
To reset the EC (embedded controller), shutdown the machine, then actuate the emergency reset hole at the bottom with a paper clip.

That's the only advice I can give you. Yours is a hardware issue which is offtopic here. Better ask at Lenovo forums.