linrunner/TLP #684: tlp causes eth0 link not to become ready (no-carrier)

dkwo, 2023-04-04:
If I install `tlp` (not even enable the runit service) on void linux (x86_64-musl), the link eth0 does not become ready. This is an ethernet cable connected to the monitor, which is connected to the dock via usb cable. If I remove the tlp package, then I see `dmesg: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready` as well as `ethtool eth0 ...	Link detected: yes`, otherwise eth0 has no carrier.

https://github.com/void-linux/void-packages/issues/43148

dkwo, 2023-04-05:
I cannot reproduce reliably, for now I can attach a tlp-stat without the issue.
I suspect it is related to tlp's udev rules.
I'm positive that tlp was causing the issue a few days ago though.

I created a private gist and mentioned you: can you see it?

linrunner, 2023-04-05:
What is actually so damn hard about answering all the questions from the template?

> 1. Does the problem occur **on battery or AC or both**?
> 4. Full output of **tlp-stat** via https://gist.github.com/ **for all matching cases of 1.** (not as file attachment, no screenshots)

dkwo, 2023-04-05:
1. it only happens when attached to the thinkpad ultra dock, power through the dock
2. https://gist.github.com/dkwo/dde36f371689e65863643d26b314f8f8

dkwo, 2023-04-05:
> I suggest to uninstall - or [disable](https://linrunner.de/tlp/faq/operation.html#how-do-i-stop-or-disable-tlp-completely) - TLP, reboot and run the ThinkPad like this for a few days to see if it reoccurs.

ok, will remove tlp and report in a week.

dkwo, 2023-04-11:
without tlp for about a week: the issue did not arise

linrunner, 2023-04-12:
Try to exclude the ethernet adapter from runtime pm

    RUNTIME_PM_DENYLIST="00:1f.6"

and reboot.

ps. deactivating the service is useless -> [FAQ: How do I stop or disable TLP completely?](https://linrunner.de/tlp/faq/operation.html#how-do-i-stop-or-disable-tlp-completely)

linrunner, 2023-04-12:
ps. Furthermore, you usually connect a monitor to this dock via HDMI or DP, since it does not have a USB-C port. A monitor on a USB-A port would be possible in principle, but then you would see a display link USB adapter in the list. So far, what you write is not plausible for me ...

dkwo, 2023-04-12:
thanks, excluding the adapter works.
the monitor is connected to that dock both via usb-a and dp cables.
i do that because sometimes i instead connect the monitor to another laptop via usb-c, and this way i do not need to unplug ethernet cables.

closed 2023-04-12 by dkwo

linrunner, 2023-04-12:
I see. Thanks for explaining.