linrunner/TLP #687: Disabled USB ports even using the correct settings

firefoxlover, 2023-04-25:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
On my new machine the default TLP configs disable all the USB ports, I tried them with a mouse that worked before and does on another computer. Now I edited the `/etc/tlp.d/00-template.conf` and added these lines:

```
# Wifi powersave
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=off # default on

PCIE_ASPM_ON_BAT=powersupersave

USB_AUTOSUSPEND=0 # default 1
USB_EXCLUDE_BTUSB=1 # default 0, bluetooth dongles
USB_EXCLUDE_PHONE=1 # default 0, prevents charging sometimes
USB_EXCLUDE_WWAN=1 # default 0
#USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN=1 #workaround if suspended devices block
```

So I have some questions.

- may it be powersupersave in PCIE ?
- does actively disabling it result in worse batterylife than not having TLP at all?

**Expected behavior**

Disabling all these settings should be the last resort for fixing it. I reloaded the conf with `sudo tlp usb`

**To Reproduce**

Get a new AMD Thinkpad and install tlp on it.

[tlp-stat](https://gist.github.com/firefoxlover/7f9bdae9046a4c89e1bf60ce1ae53336)

**Additional context**

Operating System: Fedora Linux 38
Kernel Version: 6.2.11-300.fc38.x86_64 (64-bit)
Processors: 8 × AMD Ryzen 5 PRO 3500U w/ Radeon Vega Mobile Gfx
Graphics Processor: AMD Radeon Vega 8 Graphics
Manufacturer: LENOVO
System Version: ThinkPad T495

linrunner, 2023-04-26:
Duplicate of #436 

TLP has no feature to disable ports. I suspect this is a known (kernel) issue, please use the workaround described in the FAQ: https://linrunner.de/tlp/faq/usb.html#newly-inserted-usb-devices-not-recognized-in-battery-mode

In your case the PCI addresses are different:

    RUNTIME_PM_DENYLIST="04:00.3 04:00.4"

You should remove your own workaround attempts afterwards.

Btw: I have deleted your second identical issue.

firefoxlover, 2023-04-30:
Thank you, funny T495 it is. Havent read the FAQ, will close this issue and try if it works again.

closed 2023-04-30 by firefoxlover