linrunner/TLP #690: T480 hot swap

vdbash, 2023-05-06:
I have a Lenovo T480 with ubuntu 22.04 with main(internal or BAT0) and auxiliary batteries (BAT1). When the device using the external battery (BAT1 discharging), if I remove it I expect the power source automatically switch to main battery and device stays on. But the what actually happens is that the device looses the power and shuts down. Then I can restart the laptop with internal battery. Now if I add the external battery again it does not recognized and tlp-stat tells me that only BAT0 is available.

linrunner, 2023-05-08:
Which battery is discharged is solely up to Lenovo's embedded controller (EC) firmware, TLP has no control over this.

I recommend resetting the EC using the emergency reset hole on the bottom (refer to the hardware maintenance manual). If this does not help, there is a hardware problem (battery or system board).

closed 2023-05-08 by linrunner

vdbash, 2023-05-12:
Hi
The problem is not which battery discharging, the problem is that it is not
doing a hot swap. When I remove the external battery while the internal
battery has 80% charge, the laptop shouldn't lose power and go off. Anyway,
you can consider it a bug report if you like since you already closed the
thread. I have uninstalled TLP and now I can hot swap

vdbash, 2023-05-12:
> Oh yes, we are talking about the same thing.
>
> The moment you remove the active, i.e. just discharging, external battery
> (BAT1) to "hot swap" i.e. change it, the EC must switch to the internal
> battery (BAT0) to commence discharging it and thus operate the ThinkPad.
> All this is controlled by the EC and TLP has no influence on it, because
> there is simply no API for it from Lenovo.
>
> I think it is likely that the problem will reoccur even without TLP. This
> ThinkPad generation is prone quite to EC malfunctions, see e.g.
>
>    -
>    https://linrunner.de/tlp/faq/battery.html#disabling-the-charge-thresholds-does-not-work
>    - https://bbs.archlinux.org/viewtopic.php?id=285210
>
> Of course, I can't completely rule out the unlikely possibility of a side
> effect of TLP, but I haven't the slightest idea how to narrow down your
> problem. Even after 13 years of intensive work on the subject, I still
> don't know everything about it.
>
> Good luck.

Thanks. Appreciate it.