linrunner/TLP #691: Conflict with supergfxctl

Tarballwalf, 2023-05-12:
**Is your feature request related to a problem? Please describe.**

Currently, trying to install supergfxctl with tlp over on arch, with their provided repository, results in package conflicts. I'd like to be able to control the GPU in my laptop 

[Supergfxctl](https://gitlab.com/asus-linux/supergfxctl)
[Their repository](https://asus-linux.org/wiki/arch-guide/)

Also, if it could be possible to use asusctl too

linrunner, 2023-05-12:
Package conflicts must be resolved by the respective package maintainer/provider. They are not dealt with here in the upstream issue tracker. Thanks for your understanding.

ps. I know nothing about this tool. Whether and why there is a conflict with TLP is for its developers to say.

Tarballwalf, 2023-05-12:
alright, thanks for letting me know. I'll contact the developers of said tool. It seems they are really set on using power profile daemon, which on my laptop is really not doing anything.

closed 2023-05-12 by Tarballwalf