linrunner/TLP #692: USB_EXCLUDE_PHONE setting not applied

PatrickDaG, 2023-05-14:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

To enable phone charging I've added `USB_EXLUDE_PHONE=1` to to `tlp.conf`.
Phone still does not charge and is not excluded from autosuspend

**Expected behavior**

Phones get excluded from autosuspend, charging as intended.

**To Reproduce**

Steps to reproduce the unexpected behavior:
- Add `USB_EXLUDE_PHONE=1` to `tlp.conf`
- restart tlp
- plugin phone/check `tlp-stat -u`

1. Does the problem occur on battery or AC or both? both
5. **Full output of `tlp-stat`
https://gist.github.com/PatrickDaG/606c65bb1bd48fb74310f40e12665461

linrunner, 2023-05-14:
Hi, 

what brand and model is the phone?

I don't see any USB devices at all in your tlp-stat output. Strange. Was the phone connected when you created the output?

Please show the output of (phone connected):

    lsusb -v

PatrickDaG, 2023-05-14:
It's a Samsung zflip 4.
The phone was connected when I called `tlp-stat`
All other devices I have function as expected.

The phone charges for 5 seconds, then stops. And also I guessed the line
```
Exclude phones    = disabled
```
meant the phone was not excluded from autosuspend.

https://gist.github.com/PatrickDaG/cb1f06b17c0ba1ef7f2786a8debb68b4

linrunner, 2023-05-15:
> And also I guessed the line **Exclude phones = disabled** meant the phone was not excluded from autosuspend.

Your assumption is not correct, the line only shows the TLP configuration for USB_EXLUDE_PHONE. Normally `tlp-stat` or `tlp-stat -u` lists all USB devices. But the list is missing in your output. I suspect a problem with NixOS resp. the NixOS **tlp** package. We'll have to address that problem later. 

Before that, I want to know how TLP detects the phone device. Please add the following line to your config file:

    TLP_DEBUG="usb"

Unplug the phone and plug it back in. Then show the output of

    sudo tlp-stat -T

PatrickDaG, 2023-05-16:
https://gist.github.com/PatrickDaG/41558452abca6d3a8edf24896b886c0c

linrunner, 2023-05-19:
I suggest to exclude the phone via the USB ID:

    USB_DENYLIST="04e8:6860"

Unplug the phone and plug it back in. Then show again the output of

    sudo tlp-stat -T

I still have to analyze why the logic does not recognize it as a phone.

To find out what goes wrong when listing the USB devices I need the output of

     /usr/share/tlp/tlp-usblist

PatrickDaG, 2023-05-19:
It turns out the phone is excluded from autosuspend I just miswrote the option in the config file. Writing the correct option also sets
```
Exclude phones    = disabled
```
to `enabled` as expected.
Still no devices show up in `tlp-stat -u`.  ` /usr/share/tlp/tlp-usblist` does not output anything.
`tlp-stat -T` now contains no hint of any devices
```
--- TLP 1.5.0 --------------------------------------------

May 19 18:49:08 patricknix tlp[2944]: Applying power save settings...done.
May 19 18:49:08 patricknix tlp[2944]: Setting battery charge thresholds...done.
May 20 01:23:11 patricknix tlp[123151]: Applying power save settings...done.
May 20 01:23:11 patricknix tlp[123151]: Setting battery charge thresholds...done.
May 20 01:39:46 patricknix tlp[170262]: Applying power save settings...done.
May 20 01:39:47 patricknix tlp[170262]: Setting battery charge thresholds...done.
```

linrunner, 2023-05-20:
I also overlooked your initial config typo.

Did you unplug the phone and plug it back in after changing the config? USB devices are only recognized and configured when they are plugged in (again), this creates the log line.

PatrickDaG, 2023-05-21:
I have tried replugging it both with the old config and with the fixed one. In both cases, the log did not contain any mention of a device. `tlp-usblist` is also empty in both cases.

linrunner, 2023-05-21:
Do you still have the above mentioned TLP_DEBUG entry in your config? Please show the effective config via

    tlp-stat --cdiff

linrunner, 2023-07-15:
Due to inactivity I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2023-07-15 by linrunner