linrunner/TLP #695: Wifi fails on Dell XPS 15 9500 post-resume on AC

robgssp, 2023-07-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

On AC power, wifi on my laptop fails after a suspend/resume cycle with TLP enabled.

**Expected behavior**

Wifi should work after resume.

**To Reproduce**

On boot with kernel >=5.17 and TLP enabled, suspend and resume while on AC power. Wifi will fail to connect accompanied by stack-traces from iwlwifi.

Problem occurs on AC power only; unplugging the laptop clears the issue.

Wifi will sometimes fail to connect entirely, and sometimes connect at very slow speeds - ~100kbit instead of the usual ~500mbit.

tlp output, and iwlwifi errors: https://gist.github.com/robgssp/0368b1942f601a28fe6bc7a76fb925ba

Interestingly, blacklisting the SD card drivers (rtsx_pci and rtsx_pci_sdmmc) causes the wifi to fail unconditionally, before suspend and whether on battery or AC.

**Additional context**

Laptop is a Dell XPS 15 9500 with a "Killer AX1650s" (Intel AX201) wifi card.

robgssp, 2023-07-02:
Setting WIFI_PWR_ON_AC="on" partially mitigates the issue but does not resolve it - with that set, wifi doesn't fully cut out but is stuck in "slow mode" until AC is disconnected.

linrunner, 2023-07-05:
Hi. What you are observing is not a TLP but obviouslyy a kernel issue. You may - as you have already started - change settings in TLP to work around it.I would try for example to disable wifi power save completely

    WIFI_PWR_ON_AC=""
    WIFI_PWR_ON_BAT=""

linrunner, 2023-07-26:
Due to inactivity I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2023-07-26 by linrunner