linrunner/TLP #697: Systemd hangs on Arch Linux

AkinoKaede, 2023-07-04:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When I exec `systemctl start tlp` to start tlp, it will be hangs.

**Expected behavior**

TLP can be started by systemd.

**To Reproduce**

Exec  `systemctl start tlp` 

**Additional context**

I deleted `multi-user.target` in `After` from systemd file, it was fixed.

I had a similar problem with other software. EmixamPP/linux-enable-ir-emitter#101

```
# /usr/lib/systemd/system/tlp.service

# tlp - systemd startup/shutdown service
#
# Copyright (c) 2022 Thomas Koch <linrunner at gmx.net> and others.
# This software is licensed under the GPL v2 or later.

[Unit]
Description=TLP system startup/shutdown
After=NetworkManager.service
Before=shutdown.target
Documentation=https://linrunner.de/tlp

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/bin/tlp init start
ExecReload=/usr/bin/tlp start
ExecStop=/usr/bin/tlp init stop

[Install]
WantedBy=multi-user.target
```

AkinoKaede, 2023-07-05:
这似乎是因为 systemd-udevd 引起的

AkinoKaede, 2023-07-05:
This seems to be caused by systemd-udevd

AkinoKaede, 2023-07-05:
In my view, I think  `multi-user.target` should be reached after TLP started. But TLP will be started after `multi-user.target` .

linrunner, 2023-07-05:
Hi. Works as designed. `tlp.service` should run as late as possible in the boot process, hence the `After=multi-user.target`. 

Does this happen during boot or only when you start the service manually?

Generally I require complete outputs as described in the [issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md). Provide them via https://gist.github.com/ please.
______________________________________________

closed 2023-07-05 by AkinoKaede

linrunner, 2023-07-05:
I see. Thanks for the feedback.