linrunner/TLP #701: Battery thresholds not working for Huawei Matebook

guillemap, 2023-08-10:
- [x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
- [x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

After checking many threads online couldn't find a solution, seems huawei-wmi drivers are not enabling the thresholds with the Matebook + Ubuntu 22.04 + kernel 6.2.0-26-generic
Battery thresholds are applied but the laptop charges to 100% anyway.
I have uninstalled default battery `power-profiles-daemon` from gnome and didn't change anything.

The output of `tlp-stat -b`

```
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: huawei
Supported features: charge thresholds
Driver usage:
* vendor (huawei_wmi) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0:  0(default)..99
* STOP_CHARGE_THRESH_BAT0:   1..100(default)

/sys/devices/platform/huawei-wmi/charge_control_thresholds  = 75 80 [%]

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = DYNAPACK
/sys/class/power_supply/BAT1/model_name                     = HB4593R1ECW
/sys/class/power_supply/BAT1/cycle_count                    =    317
/sys/class/power_supply/BAT1/charge_full_design             =   7410 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   6634 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   5359 [mAh]
/sys/class/power_supply/BAT1/current_now                    =    962 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

Charge                                                      =   80.8 [%]
Capacity                                                    =   89.5 [%]
```

linrunner, 2023-08-10:
You say it yourself, the thresholds are written correctly by TLP, you can see it in the output. 

I see several possible causes:
1. your model doesn't support load thresholds at all and the kernel driver writes into the void
2. the hardware or the embedded controller, which controls the loading, has a problem.
3. the kernel driver has a bug

TLP is not involved in any of these possible causes and I can't help you at this point.

linrunner, 2023-08-10:
ps. For posterity, I would like to have exact details of your hardware. You have not fulfilled the requirement for _complete_ **tlp-stat** outputs, but in this case

    tlp-stat -s

is sufficient. Thank you.

guillemap, 2023-08-10:
Sure, didn't want to bloat the issue thread ;)

Output of `tlp-stat -s`

```
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = HUAWEI M1D KPL-W0X
BIOS           = 1.24
OS Release     = Ubuntu 22.04.3 LTS
Kernel         = 6.2.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-6.2.0-26-generic root=/dev/mapper/ubuntu--vg-root ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.9)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 12:11:42,     23 sec(s) ago
Mode           = battery
Power source   = battery

Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
>>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.

Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
>>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.

```

linrunner, 2023-08-17:
Since I can't help here, I'll close. Nevertheless, you are welcome to post new findings about your problem here.

closed 2023-08-17 by linrunner