linrunner/TLP #702: Obtain the current main mode of operation (automatic/manual)

drws, 2023-08-16:
**Is your feature request related to a problem? Please describe.**

It seems there is currently no way to query the current main mode of operation (_automatic_ or _manual_). One can only infer by comparing `Mode` and `Power source` values from `tlp-stat -s`, but this is not a reliable indication for such a basic parameter.

**Describe the solution you'd like**

`tlp-stat -s` to also display the actual commanded main mode of operation.

**Describe alternatives you've considered**

As explained before, it is possible to approximate the current mode, but since it's one of the basic (input) parameters, it should have a direct, unambiguous and readily-available indication.

linrunner, 2023-08-17:
In fact your feature request was implemented back in 2018 with version 1.2 and looks like this:

> +++ TLP Status
> State          = enabled
> RDW state      = enabled
> Last run       = 15:03:34,      3 sec(s) ago
> **Mode           = battery (manual)**
> Power source   = AC

May I please see your output from **tlp-stat -s** (not only my cutout but everything)?

drws, 2023-08-18:
Main mode indication is missing in my case. The issue is present on three similar machines (two laptops, one desktop), all running latest Arch Linux with TLP v1.5.0 [from official repo](https://archlinux.org/packages/extra/any/tlp/). The requested output:

```
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = [multiple values]
BIOS           = [multiple values]
OS Release     = Arch Linux
Kernel         = 6.1.45-1-lts #1 SMP PREEMPT_DYNAMIC Fri, 11 Aug 2023 11:28:00 +0000 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-linux-lts root=/dev/sda2 rw acpi.ec_no_wakeup=1 btusb.enable_autosuspend=0
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 12:41:19,    210 sec(s) ago
Mode           = battery
Power source   = battery
```

There is no main mode indication in `Mode` value (as in your case) under any circumstance (fresh boot, automatic mode selected, manual mode selected). Please also note other differences between our outputs:
- excessive whitespace after the comma in `Last run` value (in my case)
- vertical alignment present in my case, but not yours

Also, there's a single excessive space after value of `Init system` (`"systemd "`) in my case.

linrunner, 2023-08-21:
I can't reproduce this on Arch Linux. So let's me see what happens on one of your machines by showing the output of

    sudo tlp bat
    ls -la /run/tlp
    cat /run/tlp/manual_mode

> Please also note other differences between our outputs:

Your presumed differences come from combining a code block (your output) with a quote (my example). This misses the problem.

linrunner, 2023-09-11:
> -rw-r-----  1 root root    2 Sep  4 20:49 manual_mode

Here we have the reason for your symptom. The status file is only readable by root, so the manual mode output fails when you call tlp-stat -s without sudo.

I suspect the cause is that you have an umask 0027 (or similar) set. Please show

    umask
    sudo sh -c "umask"
    sudo -i sh -c "umask"

Of course TLP has to cope with a restrictive umask. But tlp-stat should still be callable without root privilege. For TLP 1.7 I will set the rights accordingly when creating the status file.

drws, 2023-09-17:
I noticed permissions before and thought of umask so the `tlp-stat` call in my previous post was already done as root (with incorrect output). Still the changes you added for v1.7 were necessary and could be sufficient.

Also it seems that TLP should output *something* if the status is unknown, at least a warning, preferrably an error. Again, the output in question shouldn't be possible in the first place, the corrupted state should be caught, not silently skipped. This is the second issue to consider.

linrunner, 2023-09-18:
Your symptom that `tlp-stat -s` cannot read the file **manual_mode** despite `sudo` is not reproducible on my Arch Linux system. This is totally implausible. You would have to look yourself what is actually happening on your system.

The previous adjustments will be released 1.6.1. Then we'll see what you come up with.

linrunner, 2023-10-07:
I have now revised the code for detecting manual mode so that a read error cannot occur at all.

closed 2023-11-23 by linrunner

linrunner, 2025-08-04:
So here we go again ...

> However, in a special case of missing `/run/tlp/manual_mode`, `tlp-stat -s` **still** simply shows as if manual mode is disabled

That's correct. Non-existent **/run/tlp/manual_mode** means auto mode. The file is deleted when manual mode is terminated.

> Additionally, the single excessive space after value of `Init system` is still present in the output (`systemd `).

Interesting. The white space masks a problem in determining systemd version. I probably couldn't reproduce it originally because my systemd version was too old for the change at the time. See commit.