linrunner/TLP #703: intel_pstate does not work

jorgelaranjo, 2023-08-19:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
Laptop Lenovo E14 gen4 Intel has `tlp` and the following setup
```
/sys/devices/system/cpu/intel_pstate/status            = active
/sys/devices/system/cpu/intel_pstate/min_perf_pct      =  10 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      =  30 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   1
```

It would be expected that max frequency to be 30% of 4,800 which is 1,440 MHz as shown
```
/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  1410000 [kHz]
```

However a quick check of currency frequencies on CPU reports values closer to max speed without boost, of 2,600MHz
[image]

**Expected behavior**

Max cpu speed to be the same as`scaling_max_freq`

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Battery
2. Actions to reproduce the behaviour
Setup TLP and reboot

linrunner, 2023-08-20:
At the very least I expect people to provide the full output of

    sudo tlp-stat 

as **required** by the issue template. But I'll make an assumption for now.

> /sys/devices/system/cpu/intel_pstate/max_perf_pct      =  30 [%]

If `CPU_MAX_PERF_ON_BAT=30` is your actual setting, then TLP has correctly written it to the _intel_pstate_ driver as your abbreviated output shows. If _intel_pstate_ does not work as you expect it to, then you should discuss this with the _intel_pstate_ kernel developers. For the TLP issue tracker this is out of scope.

jorgelaranjo, 2023-08-20:
Closing as validated as **not** being a bug in `TLP`
[image]

closed 2023-08-20 by jorgelaranjo

jorgelaranjo, 2023-08-21:
Works as expected if `intel_pstate` in `passive` mode.
Does not work if `intel_pstate` in `active` mode, which is expected according to documentation.

Power savings seem better in active mode.