linrunner/TLP #705: Feature request

QueDemais, 2023-08-24:
**Is your feature request related to a problem? Please describe.** Hello, the problem I wanted to report is that after configuring the tlp.conf, in battery power saving mode (BAT) and maximum power performance (AC), in steam games, the fps are higher on battery (BAT), and when connected to mains (AC) the fps drops.
The cpu governor is in performance for AC and BAT.

When I used schedutil for BAT, the pc froze out of nowhere, when Google Chrome was in use, in long lives on youtube.

In tlp-stat -v, it always shows that the pc is missing drivers.

* What you want to happen
I wanted tlp to help maintain higher fps when in AC, because the maximum performance I left in AC, and better support for Intel i3 1115G4 and Intel Celeron N4000

I understand little about linux, just the basics for now, but I really like linux, and tlp linrunner and show.

Kubuntu Lts

QueDemais, 2023-08-27:
To add to the feedback:
1.5 : with these settings the errors were happening.
These settings are on i3.
On the celeron n4000 they are similar, with exceptions in the frequencies.
i3 and celeron: I noticed a very fast drain on the battery from one charge to the next, I wanted to report that too.

1.6: similar in celeron too, with the exception of the battery processor governor, which is schedutil

In 1.6 I got better and better results. 😄

I use the linrunner ppa, and with --install-recommends along with the installation

But leaving my feedback to the developer, to help with the development.

[1_5.txt](https://github.com/linrunner/TLP/files/12448074/1_5.txt)

[1_6.txt](https://github.com/linrunner/TLP/files/12448075/1_6.txt)

linrunner, 2023-08-28:
Thanks for your feedback. 

You are expecting too much in terms of the optimizations possible with TLP's default settings. TLP does _not_ include a database holding specific settings for every piece of hardware in the universe. This would be a completely different undertaking, requiring many times the resources for maintenance - if it were possible at all. Actually, this would be the job of the kernel developers, but their resources are not sufficient for this either. TLP on the other hand is a small hobby project.

So if you are not satisfied with the results of the default settings on your hardware, you must adjust them. That's exactly what you did, and that's how TLP is meant to be used.

> after configuring the tlp.conf, in battery power saving mode (BAT) and maximum power performance (AC), in steam games, the fps are higher on battery (BAT), and when connected to mains (AC) the fps drops. 

If the BAT settings give better results with the charger connected, then adopt them for AC: https://linrunner.de/tlp/settings/operation.html#set-persistent-default

> When I used **schedutil** for BAT, the pc froze out of nowhere, when Google Chrome was in use, in long lives on youtube.

That's just a kernel issue. You may use **tlp.conf** to select a different governor as a workaround or you may file a kernel bug report.

> The cpu governor is in performance for AC and BAT.
> 1.6: similar in celeron too, with the exception of the battery processor governor, which is schedutil

TLP's default settings do _not_ adjust the governor. The one you see is chosen either by the kernel of your distribution or by your own changes to **tlp.conf**.

> In tlp-stat -v, it always shows that the pc is missing drivers.
Without complete tlp-stat output, this is not actionable information.

> I noticed a very fast drain on the battery from one charge to the next, I wanted to report that too.

There is still hardware that is not supported well enough by the Linux kernel to achieve acceptable battery runtime.  Often even TLP does not help.

My conclusion: the information you have provided so far is not the kind that can lead to changes in TLP. 

However, by trying different settings you are on the right track. So if you need help to find even better settings, please ask in a Linux forum of your choice. The issue tracker is not meant as a support forum. Thanks for your understanding.

closed 2023-08-28 by linrunner