linrunner/TLP #706: Output with lack of privileges for 'tlp-stat -w'

drws, 2023-09-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When `tlp-stat -w` is run without dmesg/root privileges it prints a somewhat flawed output:

```
tlp-stat -w
--- TLP 1.6.0 --------------------------------------------

dmesg: read kernel buffer failed: Operation not permitted
No warnings detected.
```

The first issue with the output is obviously the _No warnings detected._ message which is technically correct, but could quickly mislead a less savvy user. The second issue is that behavior differs from other privilege-requiring operations, such as `tlp-stat -b`, which outputs:

```
Error: missing root privilege.
```

**Expected behavior**

All privilege-requiring operations to communicate the missing-privilege error in an uniform way.

**To Reproduce**

1. Run `tlp-stat -w` as an unprivileged user

linrunner, 2023-09-04:
Thanks for notifying. Even minor things can be improved.

Btw: It would be important to me that you answer my questions in #702.

closed 2023-09-04 by linrunner

linrunner, 2024-09-05:
Hi @drws  : TLP 1.7 Beta 1 is out with the corresponding change. Please verify -> https://github.com/linrunner/TLP/issues/760

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1:

drws, 2024-10-08:
Sorry for a late reply. It is working as expected in v1.7.0. Thank you!