linrunner/TLP #707: Battery care for MSI laptops

ElSargo, 2023-09-11:
There is a GUI program called [MControlCenter](https://github.com/dmitry-s93/MControlCenter) that provides battery care functionality such as charge limits amongst other things. However it's a qt GUI program that's only packaged for opensuse and this really seems like the kind of thing that belongs in tlp.

Could tlp implement some of the features? Is there anything I can do to help? Thanks.

linrunner, 2023-09-12:
In fact since march there is a [driver in the mainline kernel](https://git.kernel.org/pub/scm/linux/kernel/git/pdx86/platform-drivers-x86.git/commit/drivers/platform/x86/msi-ec.c?h=review-hans&id=392cacf2aa10de005e58b68a58012c0c81a100c0) seemingly implementing the standard attributes `charge_control_start/end_threshold`.

So lets see if your kernel is recent enough. Please show:

    sudo tlp-stat -s -p

ElSargo, 2023-09-12:
Here it is
```
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = Micro-Star International Co., Ltd. REV:1.0 Summit E14Evo A12M
BIOS           = E14F1IMS.11B
OS Release     = NixOS 23.05 (Stoat)
Kernel         = 6.5.1 #1-NixOS SMP PREEMPT_DYNAMIC Sat Sep  2 07:13:30 UTC 2023 x86_64
/proc/cmdline  = initrd=\efi\nixos\jc70zm4b04lmr094k75w7ird64ajwgzc-initrd-linux-6.5.1-initrd.efi init=/nix/store/x3icvkrddrz9vbzhisnpq3kg6rdlr3d1-nixos-system-SargoSummit-23.05.20230907.4f77ea6/init i915.force_probe=46a6 loglevel=4
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 19:41:32,      6 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4498 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2821 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available) 

Charge                                                      =   62.7 [%]
Capacity                                                    =   98.9 [%]
```
I'm on the latest kernel but I think NixOS doesn't have the msi-ec module installed by default.
`lsmod | rg msi-ec` no output
I didn't know about this driver before so I'll try installing it and see if it works and update this issue.

ElSargo, 2023-09-12:
The addresses for battery control in mcontrolcenter work flawlessly without msi-ec, and haven't been changed in over a year, so I am assuming that a patch for msi-ec changing the firmware id should be all thats required for the driver to function.

linrunner, 2023-09-12:
I find the approach of restricting the driver to individual BIOS versions strange. 

Before I start programming a TLP plugin for MSI, I would like to be sure that you are available for testing. Once you have managed to load the driver, I could get started.

Do you know of any restrictions on the value of the charge thresholds besides this hint in the driver commit? 
> note, that start and end thresholds control the same EC parameter and are always 10% apart

Otherwise I would assume 10 <= **charge_control_end_threshold** <= 100. Values < 10 for **charge_control_end_threshold** make no sense because of the calculation rule for **charge_control_start_threshold**.

ElSargo, 2023-09-12:
Judging from mcontrolcenter, it looks like any value from `30% <= charge_control_end_threshold <= 100%` is fair game. In the msi-ec driver the limits for the threshold are set to `0x8a <= charge_control_end_threshold  <= 0xe4` although it isn't obvious to me what percentage that is `53% - 89%`?

The driver has lists of firmware versions to check against and find valid configurations for so I think it's just a case of no-one has tested this version.

ElSargo, 2023-09-12:
OK after patching the module it loads fine and appears to be working correctly.
Tlp is also able to read the charge thresholds 
```
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = Micro-Star International Co., Ltd. REV:1.0 Summit E14Evo A12M
BIOS           = E14F1IMS.11B
OS Release     = openSUSE Tumbleweed
Kernel         = 6.5.2-1-default #1 SMP PREEMPT_DYNAMIC Fri Sep  8 07:32:27 UTC 2023 (fdde566) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.5.2-1-default root=UUID=73c5bc00-fc93-413f-98ec-ad88b2770c3c splash=silent resume=/dev/disk/by-uuid/05609abd-53d1-4a7e-a6c2-6b25e80867a3 mitigations=auto quiet security=apparmor
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 07:16:08 AM,    410 sec(s) ago
Mode           = AC
Power source   = AC

Error: tlp.service is not enabled, power saving will not apply on boot.
>>> Invoke 'systemctl enable tlp.service' to correct this!

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4496 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2798 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =     50 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =     60 [%]

Charge                                                      =   62.2 [%]
Capacity                                                    =   98.9 [%]
```

ElSargo, 2023-09-12:
And yes I will be able to test anything.
Also the driver comes with batter charge presets
|Preset|start|stop|
|---------|-------|-------|
|min|50%|60%|
|medium|70%|80%|
|max|none|100%|

as well as a "super battery" mode, could be useful if you are making a plugin for msi

linrunner, 2023-09-25:
Sorry for the late reply, I was busy with 1.6.1. 

In the meantime I have looked at the driver code. It clarifies a few things. From looking at **charge_control_threshold_show()**  and **charge_control_threshold_store()** I reckon there is only one EC register to control the charge threshold(s) and the hardware dictates stop = start + 10. From data structures like
```
static struct msi_ec_conf CONF0 __initdata = {
	.allowed_fw = ALLOWED_FW_0,
	.charge_control = {
		.address      = 0xef,
		.offset_start = 0x8a,
		.offset_end   = 0x80,
		.range_min    = 0x8a,
		.range_max    = 0xe4,
	},
```
one can also conclude that  0 <= start <= 90 and 10 <= stop <= 100

I would detect the hardware in two steps:

1. Is it an MSI laptop?
2. Does it support load thresholds? 

Does your laptop load only the _msi_ec_ module or also _msi_laptop_?

ElSargo, 2023-09-29:
The msi_laptop module appears to be for older hardware, my laptop (Summit e14) auto-loads the [msi_wmi](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/msi-wmi.c) driver along with msi-ec.

linrunner, 2023-10-05:
Hi.

I completed an initial implementation in branch [feature/bat.d-msi](https://github.com/linrunner/TLP/tree/feature/bat.d-msi). If you don't want to build a package yourself, you may just install **bat.d/55-msi** to **/usr/share/tlp/bat.d**.

I also provide an automated test **unit-tests/charge-thresholds_msi**. To run it you need to install the tool [clitest](https://github.com/aureliojargas/clitest). Please show the output of the test script.

Thanks in advance for testing.

ElSargo, 2023-10-10:
Sorry for late reply, here is the output:
```
--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = Micro-Star International Co., Ltd. REV:1.0 Summit E14Evo A12M
BIOS           = E14F1IMS.11B
OS Release     = openSUSE Tumbleweed
Kernel         = 6.5.4-1-default #1 SMP PREEMPT_DYNAMIC Wed Sep 20 05:07:04 UTC 2023 (fdd7e9e) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.5.4-1-default root=UUID=73c5bc00-fc93-413f-98ec-ad88b2770c3c splash=silent resume=/dev/disk/by-uuid/05609abd-53d1-4a7e-a6c2-6b25e80867a3 mitigations=auto quiet security=apparmor
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle] deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 05:07:32 PM, 35 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4403 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2532 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =     50 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =     60 [%]

Charge                                                      =   57.5 [%]
Capacity                                                    =   96.9 [%]
```
And the unit test
```
#1      # +++ MSI ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2      #
#3      # --- tlp start
#4      sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#5      sudo tlp start -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#6      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #6, line 16] sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
@@ -1,2 +1 @@
-Error in configuration at STOP_CHARGE_THRESH_BAT1="9": not specified, invalid or out of range (10..100). Battery skipped.
 TLP started in AC mode (auto).
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#7      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#8      sudo tlp start -- START_CHARGE_THRESH_BAT1="97" STOP_CHARGE_THRESH_BAT1="97" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#9      sudo tlp start -- START_CHARGE_THRESH_BAT1="95" STOP_CHARGE_THRESH_BAT1="96" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#10     sudo tlp start -- START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#11     #
#12     # --- tlp setcharge w/o arguments
#13     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90"
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #13, line 29] sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90"
@@ -1,3 +1 @@
-Setting temporary charge thresholds for BAT1:
-  stop  =  90
-  start =  80 (changed to meet hardware constraint)
+Error: battery charge thresholds not available.
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[... 166 lines cut ...]
#38     # --- Reset to hardware defaults
#39     sudo tlp setcharge DEF DEF
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #39, line 101] sudo tlp setcharge DEF DEF
@@ -1,3 +1 @@
-Setting temporary charge thresholds for BAT1:
-  stop  = 100
-  start =  90
+Error: battery charge thresholds not available.
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- [... cut ...]
+/sys/class/power_supply/BAT1/charge_control_end_threshold   =     60 [%]
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#42     #

FAIL: 24 of 42 tests failed
```
Many of the tests failed, do you know if this could be an error on my end? Anything that could be interfering with it?

linrunner, 2023-10-10:
Oops. Plugin does not work at all. Let's see what happens.

Please add the following line to your TLP config:

    TLP_DEBUG="bat ps run"

Show outputs:

    sudo tlp noop
    sudo tlp-stat -T
    ls -la /usr/share/tlp/bat.d/
    ls -l /sys/module/msi*

ElSargo, 2023-10-10:
I'm so sorry I had not installed the plugin correctly in the previous test, I re-ran the tests and many more of them pass now and the plugin seems to be loaded correctly.
`sudo tlp-stat -s -b`
```
--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = Micro-Star International Co., Ltd. REV:1.0 Summit E14Evo A12M
BIOS           = E14F1IMS.11B
OS Release     = openSUSE Tumbleweed
Kernel         = 6.5.4-1-default #1 SMP PREEMPT_DYNAMIC Wed Sep 20 05:07:04 UTC 2023 (fdd7e9e) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.5.4-1-default root=UUID=73c5bc00-fc93-413f-98ec-ad88b2770c3c splash=silent resume=/dev/disk/by-uuid/05609abd-53d1-4a7e-a6c2-6b25e80867a3 mitigations=auto quiet security=apparmor
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle] deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 08:36:49 PM, 242 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: msi
Supported features: charge thresholds
Driver usage:
* natacpi (msi_ec) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  don't care (hardware enforces stop - 10)
* STOP_CHARGE_THRESH_BAT0/1:   10..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4628 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2992 [mAh]
/sys/class/power_supply/BAT1/current_now                    =   2359 [mA]
/sys/class/power_supply/BAT1/status                         = Charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =     90 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =    100 [%]

Charge                                                      =   64.6 [%]
Capacity                                                    =  101.8 [%]
```
MSI unit test
```
#1      # +++ MSI ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2      #
#3      # --- tlp start
#4      sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#5      sudo tlp start -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#6      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#7      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #7, line 19] sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
@@ -1 +1,2 @@
+Error in configuration at STOP_CHARGE_THRESH_BAT1="101": not specified, invalid or out of range (10..100). Battery skipped.
 TLP started in AC mode (auto).
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#8      sudo tlp start -- START_CHARGE_THRESH_BAT1="97" STOP_CHARGE_THRESH_BAT1="97" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#9      sudo tlp start -- START_CHARGE_THRESH_BAT1="95" STOP_CHARGE_THRESH_BAT1="96" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#10     sudo tlp start -- START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#11     #
#12     # --- tlp setcharge w/o arguments
#13     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90"
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #13, line 29] sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90"
[... 195 lines cut ...]
/sys/class/power_supply/BAT1/present:1
/sys/class/power_supply/BAT1/status:Charging
grep: /sys/class/power_supply/BAT1/subsystem: Is a directory
/sys/class/power_supply/BAT1/technology:Li-ion
/sys/class/power_supply/BAT1/type:Battery
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_NAME=BAT1
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_TYPE=Battery
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_STATUS=Charging
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_PRESENT=1
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_TECHNOLOGY=Li-ion
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CYCLE_COUNT=0
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15400000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_VOLTAGE_NOW=16868000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CURRENT_NOW=2233000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CHARGE_FULL_DESIGN=4546000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CHARGE_FULL=4628000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CHARGE_NOW=3357000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CAPACITY=72
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CAPACITY_LEVEL=Normal
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_MODEL_NAME=BIF0_9
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_MANUFACTURER=MSI
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_SERIAL_NUMBER=
/sys/class/power_supply/BAT1/voltage_min_design:15400000
/sys/class/power_supply/BAT1/voltage_now:16868000
```

linrunner, 2023-10-10:
@ElSargo Thanks. That looks much better.

I fixed line #19 of the unit test. The remaining discrepancies are due to the fact that you did not use the latest version of **charge-thresholds_msi**.

Please make sure you are running with the HEAD of the branch **[feature.d/bat](https://github.com/linrunner/TLP/tree/feature/bat.d)**. Since I reordered the commit history, you will need a `git pull --force`. Then install **60-macbook** from it and test with **charge-thresholds_msi**:

    sudo tlp-stat -b
    unit-tests/charge-thresholds_msi

ElSargo, 2023-10-10:
Ok I checked it out from the latest version, it seems like the failing tests are due to the driver automatically setting the start threshold 10% bellow the end. Can confirm it is changing the thresholds, I woke up to find my battery at 100% 🥲.

`sudo tlp-stat -b`
```
                                       
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: msi
Supported features: charge thresholds
Driver usage:
* natacpi (msi_ec) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  don't care (hardware enforces stop - 10)
* STOP_CHARGE_THRESH_BAT0/1:   10..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4456 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   4456 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Full

/sys/class/power_supply/BAT1/charge_control_start_threshold =     90 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =    100 [%]

Charge                                                      =  100.0 [%]
Capacity                                                    =   98.0 [%]
```
`unit-tests/charge-thresholds_msi`
```
#1      # +++ MSI ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2      #
#3      # --- tlp start
#4      sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #4, line 12] sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
@@ -1 +1 @@
-TLP started in AC mode (auto).
+TLP started in battery mode (auto).
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#5      sudo tlp start -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #5, line 14] sudo tlp start -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
@@ -1 +1 @@
-TLP started in AC mode (auto).
+TLP started in battery mode (auto).
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#6      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #6, line 16] sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
@@ -1,2 +1,2 @@
 Error in configuration at STOP_CHARGE_THRESH_BAT1="9": not specified, invalid or out of range (10..100). Battery skipped.
-TLP started in AC mode (auto).
+TLP started in battery mode (auto).
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#7      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #7, line 19] sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
@@ -1,2 +1,2 @@
-Error in configuration at STOP_CHARGE_THRESH_BAT1="101": not specified, invalid or out of range (10..100). Battery skipped
-TLP started in AC mode (auto).
+Error in configuration at STOP_CHARGE_THRESH_BAT1="101": not specified, invalid or out of range (10..100). Battery skipped.
+TLP started in battery mode (auto).
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#8      sudo tlp start -- START_CHARGE_THRESH_BAT1="97" STOP_CHARGE_THRESH_BAT1="97" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[... 101 lines cut ...]
 Setting temporary charge thresholds for BAT1:
   stop  = 100
-  start =  90 (changed to meet hardware constraint)
+  start =  90 (due to hardware constraint)
[... cut ...]
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #39, line 102] sudo tlp setcharge DEF DEF
@@ -1,3 +1,3 @@
 Setting temporary charge thresholds for BAT1:
-  stop  = 100
-  start =  90
+  stop  = 100 (no change)
+  start =  90 (no change)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#40     # --- tlp-stat
#41     sudo tlp-stat -b | grep -E 'charge_control'
#42     #

```

linrunner, 2023-10-11:
The unit test version I provided yesterday still did not match the newest plugin. Sorry. 

Please get **unit-tests/charge-thresholds_msi** again and re-test:

    unit-tests/charge-thresholds_msi

Important: the **test must be run on AC power** or many output strings won't match.

linrunner, 2023-10-13:
Additionally, I would like to know if the MSI hardware retains the thresholds over a suspend/resume cycle. Otherwise, TLP would have to set the thresholds again after resume.

1. Please configure some thresholds in **/etc/tlp.conf** (or in your own config snippet below **/etc/tlp.d/**):

```
START_CHARGE_THRESH_BAT1=70
STOP_CHARGE_THRESH_BAT1=80
```

2. Apply them with `sudo tlp setcharge`.
3. Show the output of `sudo tlp-stat --cdiff -b`
4. Suspend and resume the machine
5. Show the output of `sudo tlp-stat -b` again

Have a nice weekend.

ElSargo, 2023-10-13:
Much Better! All tests passed Except for 1.
`unit-tests/charge-thresholds_msi`
```
#1      # +++ MSI ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2      #
#3      # --- tlp start
#4      sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
[sudo] password for root: 
#5      sudo tlp start -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#6      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#7      sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #7, line 19] sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
@@ -1,2 +1,2 @@
-Error in configuration at STOP_CHARGE_THRESH_BAT1="101": not specified, invalid or out of range (10..100). Battery skipped
+Error in configuration at STOP_CHARGE_THRESH_BAT1="101": not specified, invalid or out of range (10..100). Battery skipped.
 TLP started in AC mode (auto).
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#8      sudo tlp start -- START_CHARGE_THRESH_BAT1="97" STOP_CHARGE_THRESH_BAT1="97" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#9      sudo tlp start -- START_CHARGE_THRESH_BAT1="95" STOP_CHARGE_THRESH_BAT1="96" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#10     sudo tlp start -- START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#11     #
#12     # --- tlp setcharge w/o arguments
#13     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90"
#14     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="70" STOP_CHARGE_THRESH_BAT1="90"
#15     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="XYZZY"
#16     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="9"
#17     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101"
#18     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="97" STOP_CHARGE_THRESH_BAT1="97"
#19     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="95" STOP_CHARGE_THRESH_BAT1="96"
#20     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="95" STOP_CHARGE_THRESH_BAT1="96"
#21     sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF"
#22     sudo tlp setcharge -- NATACPI_ENABLE=0 START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF"
#23     #
#24     # --- tlp setcharge w/ arguments
#25     sudo tlp setcharge 70 90 BAT1
#26     sudo tlp setcharge 70 90
#27     sudo tlp setcharge 0 XYZZY
#28     sudo tlp setcharge 0 9
#29     sudo tlp setcharge 0 101
#30     sudo tlp setcharge 97 97
#31     sudo tlp setcharge 95 96
#32     sudo tlp setcharge 95 96
#33     sudo tlp setcharge BAT0
#34     sudo tlp setcharge 0 3 BAT0
#35     sudo tlp setcharge XYZZY ABCDE BAT0
#36     #
#37     # --- Reset to hardware defaults
#38     sudo tlp setcharge DEF DEF
#39     # --- tlp-stat
#40     sudo tlp-stat -b | grep -E 'charge_control'
#41     #

FAIL: 1 of 41 tests failed
```
Also I believe the charge thresholds are persisted indefinitely, as they are preserved after reboot or even booting another distro without any battery software at all! Here are the results just to make sure though:
`sudo tlp setcharge`
```
Setting temporary charge thresholds for BAT1:
  stop  =  80
  start =  70 (due to hardware constraint)
```
`sudo tlp-stat --cdiff -b`
```
--- TLP 1.6.1 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0542: START_CHARGE_THRESH_BAT1="70"
/etc/tlp.conf L0544: STOP_CHARGE_THRESH_BAT1="80"
/etc/tlp.conf L0587: TLP_DEBUG="bat ps run"
[... 30 lines cut ...]
Plugin: msi
Supported features: charge thresholds
Driver usage:
* natacpi (msi_ec) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  don't care (hardware enforces stop - 10)
* STOP_CHARGE_THRESH_BAT0/1:   10..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4306 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   3010 [mAh]
/sys/class/power_supply/BAT1/current_now                    =   2395 [mA]
/sys/class/power_supply/BAT1/status                         = Charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =     70 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =     80 [%]

Charge                                                      =   69.9 [%]
Capacity                                                    =   94.7 [%]
```
Appears to be working well!

linrunner, 2023-10-13:
Perfect! Thank you very much for testing this!

The deviation is just a cosmetic issue (missing dot) in the test script. Fixed.

So what about the necessary kernel module patch for your machine - will you submit it for inclusion in the mainline kernel?

ElSargo, 2023-10-13:
Yes of course, the thing is though that the driver isn't working 100%. Reading the temperature of the GPU and the speed of the right fan isn't working, which should just be a matter of changing an address the code. I plan on resolving this issue before I submit a patch, but I haven't got round to it yet. These features aren't really important though so maybe I should submit it and fix it later? Anyhow thanks a lot for all your hard work on the plugin and on the rest of this incredible project!

linrunner, 2023-10-14:
A full customization for your model is better. Finish first. 

I expect that quite a few users of the msi plugin will find that their model is not recognized by the kernel driver. I don't know yet how to deal with this support-wise.

linrunner, 2023-10-14:
Now merged into main branch

ElSargo, 2023-10-14:
If you look at the driver, you see that most of the addresses across all configurations are identical. I think a solution could be to modify it to load a generic confoguration that contains only the standard features ( including battery care ) when a specifc version isn't found. Of course the reason why it isn't like this already could be because there is no guarantee that these values won't change for future models ect, and writing wrong values to the ec can be dangerous.

linrunner, 2023-11-23:
@ElSargo I endorse the "generic" approach. Any news on the mainline kernel patch?

@ALL **More testing is welcome!**

gabrielmalaspina, 2024-02-22:
Hello,
I hope I'm not hijacking the thread, the topic seems to be generic enough to write here.
Any chance we get a patch for another MSI laptop? I'm ready to test :)
I tried adding the firmware ver. to msi-ec.c but it didn't work. (Firmware ver. is 14F1EMS1.207)
`msi_wmi` was loaded by default, `msi_laptop` was not.

Please see below:
`--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = Micro-Star International Co., Ltd. REV:1.0 Prestige 14Evo B13M
BIOS           = E14F1IMS.506
OS Release     = Ubuntu 23.10
Kernel         = 6.5.0-17-generic #17-Ubuntu SMP PREEMPT_DYNAMIC Thu Jan 11 14:01:59 UTC 2024 x86_64
/proc/cmdline  = BOOT_IMAGE=/@/boot/vmlinuz-6.5.0-17-generic root=UUID=9bf95c02-d737-4ada-be46-ba28aa81643f ro rootflags=subvol=@ quiet splash vt.handoff=7
Init system    = systemd v253 (253.5-1ubuntu6.1)
Boot mode      = UEFI
Suspend mode   = [s2idle] deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 16:56:01, 724 sec(s) ago
Mode           = battery
Power source   = battery

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4470 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2790 [mAh]
/sys/class/power_supply/BAT1/current_now                    =    363 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available) 

Charge                                                      =   62.4 [%]
Capacity                                                    =   98.3 [%]
`
Thanks a lot !

gabrielmalaspina, 2024-02-22:
Apologies, I didn't know how to post the output. I'll use "reference"? the next time

ElSargo, 2024-02-22:
I think at this point TLP already has all the required functionality to support this. @gabrielmalaspina any issues regarding firmware versions should be directed to the [msi-ec](https://github.com/BeardOverflow/msi-ec/) GitHub page as that's where the issue lies and there isn't really anything for TLP to do.

There are dozens of open issues there for various firmware versions so in the meantime I think you need to add
 ```
"14F1EMS1.207",
```
Below line 591 to add partial support for your your version, or at least to try and load it. Make sure to find the ec reset switch before you try it and be careful!

gabrielmalaspina, 2024-02-22:
Hi,
Thanks for replying. I added it after line 691? and installed it with dkms but it's still doesn't work.
I unloaded msi_wmi and loaded msi-ec manually.
`dmesg`
```shell
[ 8414.695266] ACPI: battery: new extension: msi-ec
[ 8414.695427] msi_ec: module_init

``` 
...but when I run `tlp-stat -s -b` plugin is still generic
```shell
--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = Micro-Star International Co., Ltd. REV:1.0 Prestige 14Evo B13M
BIOS           = E14F1IMS.506
OS Release     = Ubuntu 23.10
Kernel         = 6.5.0-17-generic #17-Ubuntu SMP PREEMPT_DYNAMIC Thu Jan 11 14:01:59 UTC 2024 x86_64
/proc/cmdline  = BOOT_IMAGE=/@/boot/vmlinuz-6.5.0-17-generic root=UUID=9bf95c02-d737-4ada-be46-ba28aa81643f ro rootflags=subvol=@ quiet splash vt.handoff=7
Init system    = systemd v253 (253.5-1ubuntu6.1)
Boot mode      = UEFI
Suspend mode   = [s2idle] deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 20:39:31, 324 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = MSI
/sys/class/power_supply/BAT1/model_name                     = BIF0_9
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4546 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4470 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   4425 [mAh]
/sys/class/power_supply/BAT1/current_now                    =    182 [mA]
/sys/class/power_supply/BAT1/status                         = Charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =      0 [%]

Charge                                                      =   99.0 [%]
Capacity                                                    =   98.3 [%]

``` 
thresholds are now in percentage rather than "not available", though
Thanks for your help !

ElSargo, 2024-02-25:
Can you verify that the driver is in fact working correctly without TLP? Try interacting with the driver directly with the functionality exposed under `/sys/devices/platform/msi-ec/` and verify that the charge thresholds are working.

user529, 2024-02-25:
Thanks for the tip!
I checked `/sys/devices/platform/msi-ec/` and there is no charge_control
<pre>
# ls -la /sys/devices/platform/msi-ec/ 
total 0
drwxr-xr-x  5 root root    0 Feb 25  2024 .
drwxr-xr-x 23 root root    0 Feb 25  2024 ..
drwxr-xr-x  2 root root    0 Feb 25 13:43 cpu
lrwxrwxrwx  1 root root    0 Feb 25  2024 driver -> ../../../bus/platform/drivers/msi-ec
drwxr-xr-x  2 root root    0 Feb 25 13:43 gpu
drwxr-xr-x  2 root root    0 Feb 25 13:43 power
lrwxrwxrwx  1 root root    0 Feb 25  2024 subsystem -> ../../../bus/platform
-r--r--r--  1 root root 4096 Feb 25 13:43 available_fan_modes
-r--r--r--  1 root root 4096 Feb 25 13:43 available_shift_modes
-rw-r--r--  1 root root 4096 Feb 25 13:43 battery_mode
-rw-r--r--  1 root root 4096 Feb 25 13:43 cooler_boost
-rw-r--r--  1 root root 4096 Feb 25 13:43 driver_override
-rw-r--r--  1 root root 4096 Feb 25 13:43 fan_mode
-r--r--r--  1 root root 4096 Feb 25 13:43 fw_release_date
-r--r--r--  1 root root 4096 Feb 25 13:43 fw_version
-r--r--r--  1 root root 4096 Feb 25 13:43 modalias
-rw-r--r--  1 root root 4096 Feb 25 13:43 shift_mode
-rw-r--r--  1 root root 4096 Feb 25  2024 uevent
-rw-r--r--  1 root root 4096 Feb 25 13:43 webcam
-rw-r--r--  1 root root 4096 Feb 25 13:43 webcam_block
</pre>

It seems I have to make one step back and return to deal with the msi_ec module.

linrunner, 2024-02-25:
@gabrielmalaspina @user529 You should discuss how to properly patch the kernel driver for your hardware at its repository: https://github.com/BeardOverflow/msi-ec/issues

linrunner, 2024-09-05:
Hi @ALL : TLP 1.7 Beta 1 is out with this feature. More testers wanted -> https://github.com/linrunner/TLP/issues/760

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1:

closed 2024-09-27 by linrunner

DockterTeagle, 2024-10-19:
I have tlp version 1.7.0 and a msi-ge75-raider and I have the output that charge_control_start_threshold is not available, nor is charge_control_end_threshold and i am on nixos ,is there a config I missed?

linrunner, 2024-10-23:
@DockterTeagle most likely the kernel module does not yet support your model/firmware version. As written above:

> You should discuss how to properly patch the msi_ec kernel driver for your hardware at its repository: https://github.com/BeardOverflow/msi-ec/issues