linrunner/TLP #708: TLP in LG Gram stops working after upgrade to version 1.6

RaistGH, 2023-09-14:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Since new 1.6 version in LG Gram laptop shows an error when trying `tlp setcharge` or `tlp fullcharge`: "Error: battery charge thresholds not available." and `/sys/devices/platform/lg-laptop/battery_care_limit` value is not written.

**Expected behavior**

`/sys/devices/platform/lg-laptop/battery_care_limit` to be written when executing tlp commands

**To Reproduce**

- LG Gram laptop
- Ubuntu 20.04
- Kernel 5.15.0-83-generic
- Both AC and Battery

```
sudo tlp setcharge 
Error: battery charge thresholds not available.
```
```
sudo lsmod | grep lg_laptop                                                                                                                                          
lg_laptop              24576  0
sparse_keymap          16384  1 lg_laptop
wmi                    32768  3 intel_wmi_thunderbolt,lg_laptop,wmi_bmof
```
```
sudo tlp-stat -b
--- TLP 1.6.0 --------------------------------------------

+++ Battery Care
Plugin: lg
Supported features: none available
Driver usage:
* natacpi (lg_laptop) = inactive (laptop not supported)

+++ Battery Status: CMB0
/sys/class/power_supply/CMB0/manufacturer                   =  LG 
/sys/class/power_supply/CMB0/model_name                     = LGC-LGC
/sys/class/power_supply/CMB0/cycle_count                    =    648
/sys/class/power_supply/CMB0/energy_full_design             =  72770 [mWh]
/sys/class/power_supply/CMB0/energy_full                    =  54110 [mWh]
/sys/class/power_supply/CMB0/energy_now                     =  42870 [mWh]
/sys/class/power_supply/CMB0/power_now                      =      0 [mW]
/sys/class/power_supply/CMB0/status                         = Idle

Charge                                                      =   79.2 [%]
Capacity                                                    =   74.4 [%]
```

**Additional context**

I cannot downgrade to previous TLP version. [Repository](https://launchpad.net/~linrunner/+archive/ubuntu/tlp) only shows newest version (1.6.0-2) and maximum version in Ubuntu repository is 1.3.1-2 version and support was [only added for version 1.4](https://github.com/linrunner/TLP/issues/568).
My previous working version was 1.5.0-2.

Checking [TLP documentation for LG ](https://linrunner.de/tlp/settings/bc-vendors.html#lg) it says that Kernel 5.18 and newer uses `charge_control_end_threshold` attribute instead of `battery_care_limit`, however kernel version is 5.15 and `/sys/class/power_supply/CMB0/` directory doesn't contain any file related to charge control:
```
ls /sys/class/power_supply/CMB0/
alarm           energy_full         model_name     status      voltage_min_design
capacity        energy_full_design  power          subsystem   voltage_now
capacity_level  energy_now          power_now      technology
cycle_count     hwmon2              present        type
device          manufacturer        serial_number  uevent
```

linrunner, 2023-09-15:
For your constellation the plugin _lg-legacy_ is intended. Why this does not work here we have to find out. I will first look at my code, whether I notice something. 

Please add the output of 

    tlp-stat -s

linrunner, 2023-09-15:
I have found something. Please test the new packages from [here](https://download.linrunner.de/packages/).

RaistGH, 2023-09-15:
Yes, the patch worked :)
Thanks!

Below requested outputs with working patch:

```
sudo tlp-stat -s -b
--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = LG Electronics                   0.1 14Z990-V.AA55B
BIOS           = W1ZD1250 X64
OS Release     = Ubuntu 20.04.6 LTS
Kernel         = 5.15.0-83-generic #92~20.04.1-Ubuntu SMP Mon Aug 21 14:00:49 UTC 2023 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.15.0-83-generic root=UUID=ebf88faf-c8e4-40d8-81c2-3d6b7a9cdb93 ro pci=nommconf quiet splash vt.handoff=7
Init system    = systemd v245 (245.4-4ubuntu3.22)
Boot mode      = UEFI
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 22:31:18,     77 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: lg-legacy
Supported features: charge threshold
Driver usage:
* vendor (lg_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 80(on), 100(off) -- battery care limit

/sys/devices/platform/lg-laptop/battery_care_limit          = 100 [%]

+++ Battery Status: CMB0
/sys/class/power_supply/CMB0/manufacturer                   =  LG 
/sys/class/power_supply/CMB0/model_name                     = LGC-LGC
/sys/class/power_supply/CMB0/cycle_count                    =    648
/sys/class/power_supply/CMB0/energy_full_design             =  72770 [mWh]
/sys/class/power_supply/CMB0/energy_full                    =  53400 [mWh]
/sys/class/power_supply/CMB0/energy_now                     =  14420 [mWh]
/sys/class/power_supply/CMB0/power_now                      =  25230 [mW]
/sys/class/power_supply/CMB0/status                         = Charging

Charge                                                      =   27.0 [%]
Capacity                                                    =   73.4 [%]
```

```
ls -al /sys/devices/platform/lg-laptop/
total 0
drwxr-xr-x  4 root root    0 sep 14 17:05 .
drwxr-xr-x 33 root root    0 sep 13 00:59 ..
-rw-r--r--  1 root root 4096 sep 15 22:31 battery_care_limit
lrwxrwxrwx  1 root root    0 sep 14 17:05 driver -> ../../../bus/platform/drivers/lg-laptop
-rw-r--r--  1 root root 4096 sep 14 17:05 driver_override
-rw-r--r--  1 root root 4096 sep 14 17:05 fan_mode
-rw-r--r--  1 root root 4096 sep 14 18:03 fn_lock
drwxr-xr-x  4 root root    0 sep 14 17:05 leds
-r--r--r--  1 root root 4096 sep 14 17:05 modalias
drwxr-xr-x  2 root root    0 sep 14 17:05 power
-rw-r--r--  1 root root 4096 sep 14 17:08 reader_mode
lrwxrwxrwx  1 root root    0 sep 14 17:05 subsystem -> ../../../bus/platform
-rw-r--r--  1 root root 4096 sep 14 17:05 uevent
-rw-r--r--  1 root root 4096 sep 14 17:05 usb_charge
```

linrunner, 2023-09-16:
Great. Thanks for reporting this.

linrunner, 2023-09-18:
I just released TLP 1.6.1. PPA packages are on the way.

closed 2023-09-18 by linrunner