linrunner/TLP #709: Support for i7 1365U (thinkpad T14s Gen 4)

Tuxino88, 2023-09-19:
**Is your feature request related to a problem? Please describe.**
This processor has 2 types of cores: 2 Performance-cores core and 8 Efficient-cores (obiviously with different frequency).

The frequencies on the P cores are not set.

it would be possible:
1) set frequencies for P core separately?
2) turn off the P cores?

[image]

linrunner, 2023-09-20:
You have already throttled the CPU with all available screws: 

- `platform_profile` = low-power
- `energy_performance_preference` = power
- `no_turbo` = 1

In particular you configured

- `max_perf_pct` = 20%

which means a drastic limitation of the CPU's power consumption and implicitly also limits the operating frequencies (which unfortunately the driver does not reflect in `scaling_min/max_freq`)

What purpose do you serve by additionally messing around with the frequencies?

ps. I absolutely do not understand why so many stubbornly insist on limiting frequencies although better settings been available for many years. I am now seriously considering whether to remove the frequency settings for the _intel_pstate_ altogether.

linrunner, 2023-09-20:
I see. If the _intel_pstate_ driver does not do what you expect, then you should address the issue to its developers.